ros-industrial/motoman #70: Error with Motoman SDA10F package

jcgarciaca, 2015-12-01:
Hello,
When I try the instruction `roslaunch motoman_sda10f_moveit_config moveit_planning_execution.launch sim:=true`, the next errors appear

    [ INFO] [1448860557.614217751]: Loading robot model 'motoman_sda10f'...
    [ERROR] [1448860558.119278078]: Group 'torso' is not a chain
    [ERROR] [1448860558.119350084]: Kinematics solver of type 'kdl_kinematics_plugin/KDLKinematicsPlugin' could not be initialized for group 'torso'
    [ERROR] [1448860558.119792175]: Kinematics solver could not be instantiated for joint group torso.
    [ INFO] [1448860558.126523519]: Loading robot model 'motoman_sda10f'...
    [ERROR] [1448860558.572781895]: Group 'arms' is not a chain
    [ERROR] [1448860558.572861970]: Kinematics solver of type 'kdl_kinematics_plugin/KDLKinematicsPlugin' could not be initialized for group 'arms'
    [ERROR] [1448860558.573263572]: Kinematics solver could not be instantiated for joint group arms.
    [ INFO] [1448860558.578935053]: Loading robot model 'motoman_sda10f'...
    [ERROR] [1448860559.023278249]: Group 'sda10f' is not a chain
    [ERROR] [1448860559.023357702]: Kinematics solver of type 'kdl_kinematics_plugin/KDLKinematicsPlugin' could not be initialized for group 'sda10f'
    [ERROR] [1448860559.023930325]: Kinematics solver could not be instantiated for joint group sda10f.
    [ WARN] [1448860883.486564338]: Failed to call service /get_planning_scene, have you launched move_group? at /tmp/buildd/ros-indigo-moveit-ros-planning-0.6.5-0trusty-20151112-0345/planning_scene_monitor/src/planning_scene_monitor.cpp:461
    [ERROR] [1448860575.979981301]: MoveitSimpleControllerManager: Action client not connected: sda10f/sda10f_r1_controller/joint_trajectory_action
    [ INFO] [1448860581.096676607]: MoveitSimpleControllerManager: Waiting for sda10f/sda10f_b1_controller/joint_trajectory_action to come up
    [ INFO] [1448860586.096886543]: MoveitSimpleControllerManager: Waiting for sda10f/sda10f_b1_controller/joint_trajectory_action to come up
    [ERROR] [1448860591.097130984]: MoveitSimpleControllerManager: Action client not connected: sda10f/sda10f_b1_controller/joint_trajectory_action
    [ INFO] [1448860596.225098638]: MoveitSimpleControllerManager: Waiting for sda10f/sda10f_b2_controller/joint_trajectory_action to come up
    [ INFO] [1448860601.225290651]: MoveitSimpleControllerManager: Waiting for sda10f/sda10f_b2_controller/joint_trajectory_action to come up

So, I can't plan trajectories.

I posted this question here http://answers.ros.org/question/221714/error-with-motoman-sda10f-package/

How can I solve it?

Thank you

gavanderhoorn, 2016-02-18:
@jcgarciaca: the ROS Answers question was answered. Can we close this issue?

gavanderhoorn, 2016-07-08:
Closing due to inactivity.

closed 2016-07-08 by gavanderhoorn