ros-industrial/motoman #83: Improve meshes shading

VictorLamoine, 2016-02-18:
Fix the visual meshes shading to get smooth lightning:
http://answers.ros.org/question/226831/how-do-i-make-my-meshes-appear-shaded/

Collisions meshes should not be edited.

gavanderhoorn, 2025-03-19:
Yaskawa/Motoman have actually switched to using Collada meshes for the `visual` quality, which have better shading already.

closed 2025-03-19 by gavanderhoorn

VictorLamoine, 2025-03-20:
Oh that's great 😄 I can download STEP files on Yaskawa website, is there a public repository with the updated meshes?
I will need the GP12 shortly so I can at least work on this one if required!

gavanderhoorn, 2025-03-20:
> is there a public repository with the updated meshes?

all robot support packages for ROS are still hosted in this repository.

To clarify: the current workflow at Yaskawa always uses/prefers Collada meshes for `visual` quality. That doesn't necessarily mean 5 MB+ sizes per mesh are now acceptable, so there's still some trade-off to be made between visual fidelity and file size.

> I will need the GP12 shortly so I can at least work on this one if required!

seems like that particular [support package](https://github.com/ros-industrial/motoman/tree/f392352035480df440c37e5543c5d6960017bfc7/motoman_gp12_support) was created before the switch to Collada. @EricMarcil?

EricMarcil, 2025-03-20:
I did a quick conversion of the motosim models to dae files. It's on my personal branch here:
https://github.com/EricMarcil/motoman/tree/GP12-Meshes-Update

I haven't time to test it yet. 

@gavanderhoorn Do we want to eventually update model in the main repo? Or that may cause problem to people that might already be using the model?

VictorLamoine, 2025-03-21:
https://github.com/user-attachments/assets/2cfeca3e-00f1-4346-bbc9-f8662d686b0c

the tool center geometry does not look right!

VictorLamoine, 2025-03-21:
I made my own (very) detailed version here.

[image]

EricMarcil, 2025-03-21:
@VictorLamoine Thanks for the feedback. I've updated my branch with the fix.
Yesterday, I only converted the models on Windows and didn't have chance to switch to my Ros environment on Linux to test it. Seems like I miss type the model offset by 3mm. The T-axis model in MotoSim was also badly represented, I had to pull it from another model that has the same T-axis flange.
I tested everything now and it is all working.