ros-industrial/motoman #84: SDA5F: controller is taking too long to execute

qqfly, 2016-03-09:
Hello,

**I.** I'm working with a SDA5F robot, and I can realize the motion planning function with MoveIt. However, when I executed the planned path, some problems came up: 

When I chose the single arm groups (arm_left & arm_right), the robot performed well;
But when I chose the sda10f group (for all joints of SDA5F), the robot could follow the path, but MoveIt cannot receive the _/joint_trajectory_action/result_ topic, and return the error codes: 

_Controller is taking too long to execute trajectory (...). Stopping trajectory.
Execution completed: TIMED_OUT
Transitioning goal to LOST_

and then moveit breakdown.

**II.** Besides, I have another question, Could the motoman_driver realize the Jacobian basd control (speed control, differential control or something like that)? I have tested the /joint_path_command and controlled the joint_t from _curPosition_ to _curPosition+dq_ in each iteration. the joint_t of robot could move, but it would stop in each iteration.
Or, can I direct control the joint velocity through this package?

Thanks a lot!

shaun-edwards, 2016-03-17:
@qqfly, your first issue could be related to a bug.  See PR #79 for a potential fix.

As for your second question.  At a low level, you could achieve what you describe.  The motoman driver sends incremental motion to the robot controller.  PR #16 actually implemented this, although the code is out of date by this point.

qqfly, 2016-04-07:
@shaun-edwards , The first problem was solved, following PR #79 .

As for the second one, PR #16 looks like what I want, but I haven't yet finished it.

qqfly, 2016-04-07:
BTW, I'll try PR #37 too.