ros-industrial/motoman #97: driver: trajectory not sent to controller after being rejected once

jettan, 2016-06-16:
```
[/motion_streaming_interface: motion_ctrl::MotomanMotionCtrl::setTrajMode: 86 ERROR] [1466091365.051498907]: Failed to set TrajectoryMode: Not Ready (5) : Controller in TEACH mode (5004)
[/motion_streaming_interface: joint_trajectory_streamer::MotomanJointTrajectoryStreamer::send_to_robot: 326 ERROR] [1466091365.051527535]: Failed to initialize MotoRos motion.  Trajectory ABORTED.  Correct issue and re-send trajectory.
```

When the driver is started in teach mode and we try to move the robot this error occurs as expected, but when we put it in remote mode, the driver doesn't recover and needs to be restarted.

gavanderhoorn, 2016-06-16:
Added the `bug` label since the error message implies that re-sending the trajectory should result in recovery of the driver, but it doesn't appear to be the case (according to @jettan, only restarting the involved nodes works).

gavanderhoorn, 2016-06-16:
@jettan: if you have the opportunity, a bag file with the relevant topics would be highly appreciated.

If it includes the retry (by re-sending) that would be even better.