ros-industrial/motoman #115: How to connect ROS_Industrial with FS100 controller

Jonasuno, 2016-08-22:
Dear all,

I have downloaded the ROS-package 'motoman', how to connect it with FS100 controller through MotoPlus? We have tried many times according to the manuals "Installing the Motoman FS/DX ROS Server(Indigo)" and "Motoplus-ROS Incremental Motion interface Engineering Design Specifications"
more detailed instruction would be preferred.

Thank you!

gavanderhoorn, 2016-08-22:
Can you please describe at which step you failed / what didn't work as described in the installation tutorial?

Especially *Installing the Motoman FS/DX ROS Server(Indigo)* should result in a working MotoRos installation, provided your controller meets the minimal requirements.

Jonasuno, 2016-08-22:
The motoplus file cannot upload  to FS100 controller using the Debug (telnet 10.0.0.2) function in MotoPLus IDE, It's always Login incorrent.  
How to connect  FS100 controller through MotoPlus and ROS Industrial?  I don't understand the connection between them, there are specific steps to resolve it. Thank you @gavanderhoorn ！

gavanderhoorn, 2016-08-22:
Are you building the MotoRos application from source? If you don't need to modify it, a simple copy of the provided binary ([here](https://github.com/ros-industrial/motoman/tree/jade-devel/motoman_driver/MotoPlus/output/FS100/MotoRosFS100)) would probably be easier.

In any case: is using a usb stick not an option for you?

@ted-miller

ted-miller, 2016-08-22:
The tutorial page (_Installing the Motoman FS/DX ROS Server_) explains that you must load the MotoPlus file using the programming pendant.  You cannot load the file over the telnet connection.

Section 2.1 explains how to load the file.  Which step are you having trouble with?
"_Locate the correct binary for your robot controller and place this file on an external memory card; Compact Flash (CF) or USB. Insert the memory card into the robot's programming pendant. Turn on the robot controller while holding the [Main Menu] key on the keypad to enter Maintenance Mode. You may release the key when you see the logo appear on the screen.

Once in Maintenance Mode, upgrade to MANAGEMENT security level by touching [System] > [Security]. (Default password is all 9's). Touch [MotoPlus APL] > [Device] to select either CF or USB memory type. Touch [MotoPlus APL] > [Load (User App)] to select and load MotoRosXXXXX.out. Touch [MotoPlus APL] > [File List] to verify that MotoROS was properly installed and no other MotoPlus applications are currently loaded on the controller._"

Please also notice the pre-requisites listed on the motoman_driver wiki page.  You must have some robot parameters enabled to load the MotoROS application onto the FS100 controller.  If you have not ordered 167352-1 (MotoROS Runtime), then please contact techsupport@motoman.com.  They will assist you with the order and enable all required parameters on your robot controller.

ted-miller, 2016-08-23:
You can also reference our user-manual which includes pictures on how to load a MotoPlus application.
Go to http://www.motoman.com/manuals/ and search for "166687-1CD".

gavanderhoorn, 2016-09-17:
Closing due to inactivity. 

@Jonasuno: if you are still having trouble installing MotoROS, could you provide some more detail on what is going wrong exactly and answer @ted-miller's questions? Then re-open.

closed 2016-09-17 by gavanderhoorn