ros-industrial/motoman #122: mh5: incorrect link_t position

gavanderhoorn, 2016-08-24:
It would seem `link_t` in the MH5 xacro macro is currently not at the correct position:

[image]

It looks like this is a [transform issue](https://github.com/ros-industrial/motoman/blob/605ae9989036a925195c75b00846152c145ac5a1/motoman_mh5_support/urdf/mh5_macro.xacro#L164), and not related to the origin of the mesh.

shaun-edwards, 2017-05-04:
@geoffreychiou, could you take a look at this one next week?

gavanderhoorn, 2017-06-05:
This should be fixed with the merge of #173.

closed 2017-06-05 by gavanderhoorn

shaun-edwards, 2017-06-05:
Actually it was only half fixed...now fully fixed in #175