ros-industrial/motoman #127: connect motoman controller FS100 with MotoROS

hustfzt2, 2016-08-31:
hello, all.
I am making motoman controller FS100 with MotoROS, acoording to the instruction 'http://wiki.ros.org/motoman_driver/Tutorials'.
when i advanced this step 'Installing INFORM Code', Touch [Ext Memory] > [Load] > "Job" to select and load INIT_ROS, the menu Job is not displayed in programming pendent.
and next, i have run the following command:
$ roslaunch motoman_driver robot_interface_streaming_fs100.launch robot_ip:=10.0.0.2
it fails as:
[ERROR] [1472633015.940108227]: Failed to find topic_list parameter
[ERROR] [1472633015.951909645]: Failed to find topic_list parameter
[ERROR] [1472633018.944927068]: Failed to connect to server, rc: -1. Error: 'No route to host' (errno: 113)
[ERROR] [1472633018.944939960]: Failed to connect to server, rc: -1. Error: 'No route to host' (errno: 113)
[ERROR] [1472633021.944906993]: Failed to connect to server, rc: -1. Error: 'No route to host' (errno: 113)
[ERROR] [1472633021.944911248]: Failed to connect to server, rc: -1. Error: 'No route to host' (errno: 113)
[ERROR] [1472633021.945081730]: Failed to receive message length
[ERROR] [1472633021.945151508]: Failed to receive incoming message
[ERROR] [1472633022.195100778]: Timeout connecting to robot controller.  Send new motion command to retry.

how to solve it? Thank you

gavanderhoorn, 2016-08-31:
> next, i have run the following command: [..] it fails as: [..]

If you can't load the Inform job, there is no point in continuing with the installation tutorial and / or trying to connect to the robot, as it won't work.

> Failed to connect to server, rc: -1. Error: 'No route to host' (errno: 113)

ignoring the fact that the installation is incomplete: can you `ping` the controller at the IP you passed to `robot_interface_streaming_fs100.launch`? The error message seems to suggest `10.0.0.2` is not on your local network.

gavanderhoorn, 2016-09-06:
Somehow I don't seem to have received a notification about your comment. My apologies for the late reply.

@ted-miller: could you think of a reason why the file would not be shown?

@hustfzt2: just making sure: you did copy the [relevant files](https://github.com/ros-industrial/motoman/tree/jade-devel/motoman_driver/Inform/FS100/single_arm) to your USB stick first, right?

ted-miller, 2016-09-06:
@hustfzt2  
1) When you load install the MotoPlus application, you must be in Maintenance mode.  However, you cannot load job files when you are in Maintenance mode.  So, please reboot the robot into Normal mode to load your job file.

2) Perhaps it is looking for the wrong memory card.  Touch [Ext memory] > [Device].  Then you can select either Compact Flash or USB.

3) If the files are in a folder on your memory card, touch [Ext Memory] > [Folder] to select the folder which contains the job file.

If that does not resolve your issue, please take some pictures of your pendant so that we can see your menus.

gavanderhoorn, 2016-09-16:
@hustfzt2: have you been able to install MotoROS?

gavanderhoorn, 2016-09-17:
Closing this due to inactivity.

@hustfzt2: if you still cannot install MotoROS, please provide more detail and answer @ted-miller's questions and re-open.

closed 2016-09-17 by gavanderhoorn