ros-industrial/motoman #129: move_to_joint.py not included in debians

shaun-edwards, 2016-09-06:
The `move_to_joint.py` (https://github.com/ros-industrial/motoman/blob/jade-devel/motoman_driver/src/move_to_joint.py) script is not accessible from the debian build of indigo.

The install of the `CMakeLists.txt` probably needs to be updated.

shaun-edwards, 2016-09-06:
Thanks @pbeeson for reporting.

gavanderhoorn, 2016-09-06:
Would we want to include that? It's really a debug script, and those are often not included in released versions.

shaun-edwards, 2016-09-06:
I think so.  Universal has a similar script that I believe is included in the install.  

@pbeeson is using it to verify motion, and I think others would find it useful to move the robot without bringing up moveit (or having to install MoveIt for that matter).

gavanderhoorn, 2016-09-06:
Well, last time I checked, the script doesn't work (properly) in case multiple motion groups are configured. So it would seem nice if that is fixed before distributing it.

gavanderhoorn, 2016-09-06:
Can't test it right now, and it probably doesn't make sense to delay fixing this issue. Will need to investigate multi-group compatibility later.

gavanderhoorn, 2016-09-16:
I've labelled this *bug*, even though I think this was actually purposefully left out of the binary distribution.

gavanderhoorn, 2016-11-29:
Closed in 259b468ea17fa680e68a0b91f57a2b58fa99b9bb.

gavanderhoorn, 2016-11-29:
Note that this has not made it to `jade-devel` yet.

commit 259b468e (2016-10-21): driver: install move_to_joint test script. Fix #129.
closed 2016-11-29 by gavanderhoorn