ros-industrial/motoman #134: some problems by sending trajectory goals to motoman_driver

longhuicai, 2016-10-29:
problem description

1. While using actionlib::SimpleActionClient<control_msgs::FollowJointTrajectoryAction>
,it is okay to put several points in one control_msgs::FollowJointTrajectoryGoal
. After sendGoal, the motoman mh5 manipulator works fine. 
But if I seperate these points into several control_msgs::FollowJointTrajectoryGoal in a loop to send goal, it will stop for a few time at every end of the trajectory, so the mh5 moves unsmooth.
I want to know if there is anyway to keep it move smooth when I keep sending goals in a loop.

2. According to the motoman_driver, control_msgs::FollowJointTrajectoryGoal.trajectory.points[0] should be the current joint state value. But after I use subscribe or WaitforMessage method in the loop to get the current value, the motion interface node still report this error occasionally: [ERROR] [1477567033.103768266]: Aborting Trajectory.  Failed to send point (#0): Invalid message (3) : Trajectory start position doesn't match current robot position (3011)

I want to know why. Or is there a better way to get the current joint state?

3. The communication between ROS and motomanmh5 takes too much time. To get the current state, 25 ms for State update delay is needed. To execute the trajectory goal, at least 50ms is needed according to my test. If adding the time cost of other operations, each trajectory loop step costs more than 100s. That is too long for our work. Is there any way to reduce the time cost here in ROS?

shaun-edwards, 2016-10-31:
> I want to know if there is anyway to keep it move smooth when I keep sending goals in a loop.
The follow joint trajectory action assumes that the trajectory starts and stops at zero veracity.  To achieve your desired behavior, you would need to do modify the driver node directly, like this: #37.

> But after I use subscribe or WaitforMessage method in the loop to get the current value, the motion interface node still report this error occasionally: [ERROR] [1477567033.103768266]: Aborting [...]

I have a feeling this may be some synchronization issues between your node that receives the current state and the driver/action node that receives the same message.  This could explain why you have having intermittent issues.

>Is there any way to reduce the time cost here in ROS?
If you need the best timing possible, my recommendation is to rewrite/refactor the driver/state nodes.  This will give you the best possible performance as you will be interacting at the raw socket level.  Note, that there will still be socket comms delays as well as controller latencies that will always be there.

gavanderhoorn, 2016-11-06:
To clarify some of the other observations that you report:

> But if I seperate these points into several control_msgs::FollowJointTrajectoryGoal in a loop to send goal, it will stop for a few time at every end of the trajectory, so the mh5 moves unsmooth.

the driver is designed to work with *trajectories* at the lowest level of the ROS API (ie: the `FollowJointTrajectory` server). Trajectories have a zero-velocity begin and end state, meaning that the robot will *always* come to a full stop after completing trajectory execution. This results in the behaviour that you describe.

> To get the current state, 25 ms for State update delay is needed.

As you can see in [MotoPlus/StateServer.h](https://github.com/ros-industrial/motoman/blob/a98bae036230c4bf32d95af8bbf5e278ad560b67/motoman_driver/MotoPlus/StateServer.h#L35), the minimum period for the `StateServer` task is 25 ms (if `mpGetRtc()` returns `1`). You could change this, but the current value is a compromise between controller resource usage (ie: CPU), responsiveness and motion performance.

I'm not sure what the impact of changing that constant would be (other than changing the time `mpDelayTask(..)` delays the `StateServer`), perhaps @ted-miller could comment on that.

ted-miller, 2016-11-07:
Lowering the constant will increase network traffic and slightly increase CPU load.  But, I wouldn't worry too much about increasing the CPU load.  That thread is running at "normal" priority, which is pretty low relative to everything else in the controller.   Since the state-server is polling feedback-position, you should be able to take the constant as low as "1".  Given the thread priority, you're not guaranteed to get a message every ms... but you should come really close.

gavanderhoorn, 2017-02-27:
@longhuicai: if you feel we've answered your questions, could I ask you to close this issue?

PS: in the future, could you ask these kind of questions on [ROS Answers](http://answers.ros.org)? The issue tracker is really more for reporting bugs or enhancement proposals.

longhuicai, 2017-03-01:
Thanks for all the replies here. 
The problem is solved by using On-the-fly point streaming mentioned in #37.
If there is any further question, i will ask on ROS Answers.
Thanks again.

closed 2017-03-01 by longhuicai