ros-industrial/motoman #137: sia10f: no motion possible

jannho, 2016-12-02:
I am trying to move the robot in RViz and get the following errors:

`[ERROR] [1480688733.818253805]: Socket sendBytes failed, rc: -1. Error: 'Broken pipe' (errno: 32)`
`[ERROR] [1480688733.818307110]: Failed to send MotionCtrl message`
`[ERROR] [1480688733.818339549]: Failed to send CHECK_MOTION_READY command`
`[ERROR] [1480688733.818413947]: Failed to send MotionCtrl message`
`[ERROR] [1480688733.818441119]: Failed to send TRAJ_MODE command`
`[ERROR] [1480688733.818462671]: Failed to initialize MotoRos motion.  Trajectory ABORTED.  Correct issue and re-send trajectory.`
`[ERROR] [1480688735.378687546]: Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 1.561081 seconds). Stopping trajectory.`

Further information: 

I followed the tutorial on the ROS wiki for installing the Motoman FS ROS server and no errors appeared on the control panel.
The joint_state topic publishes the position of the robot correctly but the robot_status seems to be a problem: `header: 
  seq: 30861
  stamp: 
    secs: 1480689045
    nsecs: 707742047
  frame_id: ''
mode: 
  val: 2
e_stopped: 
  val: 0
drives_powered: 
  val: 0
motion_possible: 
  val: 0
in_motion: 
  val: 0
in_error: 
  val: 0
error_code: 0`

What could be the reason that no motion is possible?

gavanderhoorn, 2016-12-02:
Can you tell us how you start the involved launch files? Which parameters do you use?

jannho, 2016-12-02:
roslaunch sia10f_moveit_config moveit_planning_execution.launch sim:=false robot_ip:=10.0.0.2

Do you need the launch file too? I just use the template found in the Create a MoveIt Package for an Industrial Robot tutorial.

gavanderhoorn, 2016-12-02:
The error message (`Socket sendBytes failed, rc: -1. Error: 'Broken pipe' (errno: 32)`) seems to suggest that the driver does not have a / has lost the connection to the controller when it tries to send a trajectory.

As you say you do have `JointState`s and `RobotStatus` msgs, it would seem that it is at least partially working (and that your controller is really at `10.0.0.2`). 

Can you provide some more information on your setup? Which version of ROS are you using, which version of the driver have you installed (ie: downloaded from the github repository or installed using `apt-get`), etc.

shaun-edwards, 2016-12-02:
Are you using a wired or wireless connection between your PC and the robot?

ted-miller, 2016-12-02:
Please ensure that you have ordered part# 167352-1 from Yaskawa to ensure your robot controller is configured to fully support the MotoROS driver.

See section 2.0 (Requirements) of this page: http://wiki.ros.org/motoman_driver

jannho, 2016-12-07:
Sorry for the late response. 

I am using ROS Indigo on Ubuntu 14.04 and installed the indigo version of the driver downloading the github repository. The system verison is 3.27 
I am using a wired connection.

I ordered the part# 167352-1, but I will doublecheck if the controller is fully configured.

jannho, 2016-12-08:
The problem is solved. It turned out that the controller was not fully configured. Yaskawa updated my parameters and now everything is working fine. Thank you guys

closed 2016-12-08 by jannho

gavanderhoorn, 2016-12-08:
Thanks for reporting back.

Good to hear you got it to work.