ros-industrial/motoman #138: driver/MotoPlus: bHasPreviousData set but never used?

gavanderhoorn, 2016-12-21:
The variable `bHasPreviousData` introduced in d23afc85ed5cd11b5c3648241e75e798862b0580 seems to be set ([here](https://github.com/ros-industrial/motoman/blob/indigo-devel/motoman_driver/MotoPlus/MotionServer.c#L308) and [here](https://github.com/ros-industrial/motoman/blob/indigo-devel/motoman_driver/MotoPlus/MotionServer.c#L323)), but is then never used anywhere as far as I can tell.

Assuming the "*Allow for multiple packets to be received in single Recv command*" line in the commit message of d23afc85ed5cd11b5c3648241e75e798862b0580 is actually related, should `bHasPreviousData` be included in any of the conditionals in `MotionServer::Ros_MotionServer_WaitForSimpleMsg(..)`?

ted-miller, 2016-12-28:
You are correct.  I missed a line in a previous merge.  I'll fix it this afternoon.  But, before I try another commit, can you review my previous one so that I don't screw it up further...?

[image]

ted-miller, 2016-12-28:
This has been corrected in PR 140.

gavanderhoorn, 2016-12-28:
Thanks for the fix, but let's keep this open until #140 has been merged.

gavanderhoorn, 2016-12-30:
Fixed in 40896603425f73b98b2b3b3631d5b367c61fae4d.

closed 2016-12-30 by gavanderhoorn