ros-industrial/motoman #143: MotoROS ALARM 8003

BlazingForests, 2017-01-09:
Hi, guys

I am following this wiki http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer to setup

When i finish install MotoROS, then POWER OFF and POWER ON the controller,
Boot into normal operation mode, before i load INIT_ROS

Then an ALARM comming

```
ALARM 8003
  MotoROS Cfg: Set RS000=2
    [1]
```

I think maybe the INIT_ROS, so i finish the installing INIT_ROS (ref #142)

Then boot into normal operation mode( REMOTE mode) 

But the ALARM still there

THX

gavanderhoorn, 2017-01-09:
These alarms were added very recently (#140), and it seems like they're showing their worth already.

Has your DX200 been set up by Yaskawa for MotoROS? If not, that is what the error is telling you.

gavanderhoorn, 2017-01-09:
We haven't yet had time to document them. I just created an issue for that (#144).

gavanderhoorn, 2017-01-09:
As to your problem: there are some parameters you can set yourself, but others are not user-configurable. I strongly recommend you get in touch with your Yaskawa support office and ask them for support in setting up MotoROS on your DX200.

Do you have part nr 167352-1 on your controller?

BlazingForests, 2017-01-09:
Hi, @gavanderhoorn  Sorry, What is "nr 167352-1" ??

BlazingForests, 2017-01-09:
Hi, @gavanderhoorn  I am setup DX200 for motoplus myself.

gavanderhoorn, 2017-01-09:
@BlazingForests wrote:
> Hi, @gavanderhoorn  Sorry, What is "nr 167352-1" ??

Have you read the [Requirements](http://wiki.ros.org/motoman_driver#Requirements) section of the `motoman_driver` wiki page?

BlazingForests, 2017-01-09:
Hi, @gavanderhoorn  

So, the "167352-1" is provide by ROS-Industrial or MOTOMAN？ 
"167352-1" and MotoROS app (*.out)  is not the same thing?

THX

gavanderhoorn, 2017-01-09:
@BlazingForests wrote:
> So, the "167352-1" is provide by ROS-Industrial or MOTOMAN？ "167352-1" and MotoROS app (*.out) is not the same thing?

The application is part of  `167352-1`, but Yaskawa will also change the value of a number of internal parameters for you, so you can be sure the driver works correctly. As stated on the `motoman_driver` wiki page (emphasis mine):

> The MotoROS application and source is available for public download. **However, ordering 167352-1 will ensure your controller is updated with correct system software, the MotoROS driver is installed, and all internal parameters are properly configured.**

If you didn't order `167352-1`, then you can send Yaskawa a mail asking for assistance in setting the controller up for MotoROS. You send them a cmos image, they change the required parameters for you and send you the image back. After that, everything should work.

BlazingForests, 2017-01-09:
@gavanderhoorn 

Thank you very much!

So how to create the "cmos image“？

THX

gavanderhoorn, 2017-01-09:
@BlazingForests wrote:
> So how to create the "cmos image“？

Please refer to the appropriate Yaskawa manuals for that, or ask your local Yaskawa support office.

BlazingForests, 2017-01-09:
@gavanderhoorn 
OK, Thank you very much!

ted-miller, 2017-01-09:
You can save a CMOS.BIN file in Maintenance mode.  This is a backup of your robot controller configuration and user memory.

Boot your controller holding [Main Menu].  
Touch [System] > [Security] and upgrade to Management security level.
Touch [Ex Memory] > [Device] > Select either CompactFlash or USB.
Touch [Ex Memory] > [Save] > "CMOS".

BlazingForests, 2017-01-09:
@ted-miller 
Thank you very much.
I already save it, and e-mail to motoman.

THX

gavanderhoorn, 2017-01-09:
@BlazingForests: I'm closing this as the reason for the `8003` alarms is now clear (controller is not setup correctly). If you still have problems after you get your DX200 configured properly, please open a new issue.

gavanderhoorn, 2017-01-09:
Thanks for reporting the issue btw 👍. Your case was exactly why we added the runtime checks.

closed 2017-01-09 by gavanderhoorn

BlazingForests, 2017-01-10:
@gavanderhoorn 
Thank you very much!

gavanderhoorn, 2017-01-13:
@BlazingForests: we've updated the documentation to include information on the error codes you encountered. See [motoman_driver/Troubleshooting](http://wiki.ros.org/motoman_driver/Troubleshooting).