ros-industrial/motoman #144: doc: add documentation on new MotoROS ALARM codes

gavanderhoorn, 2017-01-09:
#140 added the MotoROS alarm codes (`8003`), but these have not been documented anywhere yet.

The wiki and / or the `README` (of `motoman_driver`) should be extended to provide information on them.

gavanderhoorn, 2017-01-09:
Perhaps something like the [Troubleshooting](http://wiki.ros.org/fanuc_driver/Troubleshooting) page the `fanuc_driver` has.

gavanderhoorn, 2017-01-09:
Related: #143.

ted-miller, 2017-01-09:
It's on my todo list.

Also, I now realize that I display the alarms in the wrong order.  If they don't have any parameters configured, they'll get alarms for all of the user-configurable parameters first.  Then, after rebooting like 6 times, they'll finally get an alarm that says they are dead and need to contact Yaskawa.

I need to update the SetupValidation.__lib files to return the critical error codes first.

gavanderhoorn, 2017-01-09:
> I need to update the SetupValidation.__lib files to return the critical error codes first.

Yes, that would probably avoid some user frustration.

gavanderhoorn, 2017-01-09:
I've created #145 to track the parameter checking order @ted-miller.

ted-miller, 2017-01-12:
http://wiki.ros.org/motoman_driver/Troubleshooting

gavanderhoorn, 2017-01-12:
Nice. Looks good.

closed 2017-01-12 by gavanderhoorn

gavanderhoorn, 2017-01-13:
I've changed the formatting of the page slightly ([motoman_driver/Troubleshooting?action=diff&rev2=4&rev1=3](http://wiki.ros.org/action/info/motoman_driver/Troubleshooting?action=diff&rev2=4&rev1=3)), let me know if this is acceptable (we can revert it easily).