ros-industrial/motoman #146: driver: document potential problems with line-endings in Inform jobs

gavanderhoorn, 2017-01-09:
As commented by @ted-miller in https://github.com/ros-industrial/motoman/issues/142#issuecomment-271280846:

> However, the most common reason for the error: *ERROR 3200 : NOP or END instruction not found* 
is because of the line-endings.
> 
> The robot controller requires Windows-style line endings: <carriage return><line feed>
> 
> If you open the job in a Linux text editor (like gedit), it can change the line endings to eliminate the <carriage return>.  This will cause the Error 3200.

This should be documented somewhere, probably the installation tutorial, or a *Troubleshooting* page (to be created, #144).

gavanderhoorn, 2017-01-09:
Perhaps we should provide MotoROS distributions on the wiki, packaging the application binaries with the various Inform jobs and have users download and install those instead of cloning the github repository.

gavanderhoorn, 2017-01-09:
Documenting is nice, but it's probably easier if we let the VCS deal with this. See #147.

ted-miller, 2017-01-09:
I agree with automating it.  But, I've had some people try to modify the job in GEDIT before.  So I plan to also put this information on the wiki.

gavanderhoorn, 2017-01-09:
#146 just tries to make sure that Inform jobs don't get checkout out with their line-endings already converted to Unix. If that does happen, editing with Gedit doesn't matter any more.

Good thing to document it though. [This page](https://ubuntugenius.wordpress.com/2010/10/26/creating-windows-text-files-in-ubuntu-how-to-save-as-ms-dos-end-of-line-in-gedit/) seems to show a way to keep Windows line-endings in the file when saving.

gavanderhoorn, 2017-01-13:
The newly added *Troubleshooting* page contains a sub section that covers the issue with line-endings ([here](http://wiki.ros.org/motoman_driver/Troubleshooting#ERROR_3200:_NOP_or_END_instruction_not_found)). Additionally, #146 was merged, which forces git to always use Windows line-endings when cloning the repository.

I feel this sufficiently handles the reported issue.

closed 2017-01-13 by gavanderhoorn