ros-industrial/motoman #152: Add support for "sticky" HALT

pbeeson, 2017-03-06:
For safety purposes, we would like to send a stop to our robot, and have the driver ignore all other incoming requests ("pausing all robot motion") until another service call comes in to "unpause" the robot.

It looks to me like as the current driver is written, either a stop_motion service or an empty trajectory on joint_path_command will stop the robot, but if another message comes in afterwards, the robot will simply execute the next command.  Obviously this is ideal, but I think having a secondary "sticky" "Halt" (or pause) as different from just Stop is useful for safety.

1) Does this already exist and I'm just missing something?
2) This is pretty straightforward to implement with a new service to toggle halt/unhalt, and a bool that's checked when new goals come in, so any reason why this wouldn't be a feature others would want?

pbeeson, 2017-03-06:
Actually, the more I think about this, the more I wonder if it shouldn't just go into the industrial_client so that the behavior happens for all ROS-I robots/simulators.

shaun-edwards, 2017-03-07:
I agree the proper place is the ROS-I client.  However, we will still need to implement it twice due to the replication of code in the Motoman package (not important to this issue).

Having and enable(unhalt)/disable(halt) service would better adhere to industrial norms.  However, I have always felt requiring an enable, followed by a start/motion command was not very intuitive for the user.  What you are describing is a compromise, where the robot driver is enabled automatically on startup, but could optionally be disabled/halted via ROS service.  I'd be supportive of this.

As an aside, I've been working on an industry standard [state machine](https://github.com/shaun-edwards/packml/pull/2) implementation which would achieve the same behavior.  My original intent was to apply the state machine to top-level nodes, but @phvass has a vision of all nodes, including drivers, adhering to the state machine.

pbeeson, 2017-03-07:
Cool.  We have our own FSM stuff we've been using.

It was trivial to add a /pause_robot service with a SetBool service message
that flips a bool to determine whether industrial_robot-client accepts
goals or not (and calls cancelCB on the current goal if one is running).
You are right that this would need to be added to the actionlib client in
motoman if people are using that, but it's a minimal amount of code.  I'll
send out a PR for industrial_robot_client package tomorrow, and the diff
could probably just be applied as a patch into the motoman repo, it's that
trivial.

BTW, it's a bit confusing following the motoman code through the second
implementation of industrial_robot_client that isn't actually spawned as a
client, but is used server side.  IS that expected to merge with the main
industrial_robot_client package in the future?

shaun-edwards, 2017-03-07:
>BTW, it's a bit confusing following the motoman code through the second
implementation of industrial_robot_client that isn't actually spawned as a
client, but is used server side.  IS that expected to merge with the main
industrial_robot_client package in the future?

It's on the TODO list, but it's not a high priority.  The reason for the "fork" is that the Motoman implementation supports dual arms.  The "better" solution is to bring better support for dual arms to the `industrial_robot_client` and then replace the motoman "fork".

gavanderhoorn, 2017-03-07:
> The "better" solution is to bring better support for dual arms to the `industrial_robot_client` and then replace the motoman "fork".

the "best" solution would probably be to make MotoROS IRCv2 compatible and use that.

shaun-edwards, 2017-03-07:
@gavanderhoorn, that is what I intended.

pbeeson, 2017-03-07:
PR for industrial_robot_client here: https://github.com/ros-industrial/industrial_core/pull/174
PR for motoman_driver here: https://github.com/ros-industrial/motoman/pull/153

As Gjis has mentioned, "pause" may not be the semantically correct term here.  I'm fine with something better describing the system, and I can easily replace and push a new commit that will update the PRs.

pbeeson, 2017-03-08:
@gavanderhoorn Would "Halt" work over "Pause"?

shaun-edwards, 2017-03-08:
I like enable/disable...this is the terminology robot teach pendants use.

pbeeson, 2017-03-08:
Agreed.  I will update both repos, and the PRs should reflect those changes shortly.

shaun-edwards, 2017-03-12:
I'm hoping to get to this early this week.  Sorry I've been a be slow in responding.  I really appreciate your contributions.

gavanderhoorn, 2017-05-20:
I think this has been addressed with #155. Closing, but @pbeeson please re-open if you feel that is not the case.

closed 2017-05-20 by gavanderhoorn