ros-industrial/motoman #154: Current Motoman driver behavior is unsafe

pbeeson, 2017-03-15:
We found out today, in a situation that could have turned ugly, that the default Motoman driver behavior is unsafe.  The Motoman driver automatically reengages the servos and starts sending motion commands after an estop without any acknowledgement by the user. The scenario is easy to reproduce and quite dangerous.

Run the Motoman ROS-Init program. Start a process that sends a trajectory to the Motoman driver and move the robot.  Estop the robot on the teach pendant.  Pull up the estop.  Send a new trajectory.  The robot will turn on and move.  

This is not considered safe by anyone installing robots in a production environment.  Anyone could walk up and toggle an estop button while a person is near the robot and it would start moving again.  After an estop, the driver needs to ignore all incoming commands until the user explicitly sends a command to start accepting new commands again.

I'm not sure if this is an issue with the actual ROS side doing too much in each send cycle, or on the Motoplus side doing too much on a send command, but obviously trying to engage the servos and starting motion on every incoming motion command is not a safe way to do things.

I will investigate more in the coming days.

ted-miller, 2017-03-15:
@gavanderhoorn 
This situation is perfectly analogous to a production robot installation.  Picture a traditional robot cell where the operator interacts with HMI panel (the pendant is not the primary point of control).  The operator hits an estop.  Then, to restart the robot job, the operator must push the [Start] button on the HMI.  This [Start] button has circuitry/logic to ensure that all safety sensors are in a "good" state and that the operator's entry gate is closed/locked.  Only after these checks will the HMI instruct the robot to start the job.

So, in the context of the ROS-I environment, the PC is the operator's interface (HMI).  After an estop, the operator should be forced to explicitly send a new [Start] command before motion takes off.  The PC should not just automatically send new trajectories when the estop-state is cleared.  This function that waits for the operator input should also accept some flag(s) that indicate the state of any safety sensors.  It would inevitably be the system-integrator's responsibility to define what signals the flag(s) are tied to.

From the MotoPlus side, the estop will end the robot job.  Additionally, it will reply ROS_RESULT_NOT_READY to any incoming trajectory commands.  The MotoPlus application will not restart the robot job until the PC sends a new ROS_CMD_START_TRAJ_MODE command.  Assuming the estop has been cleared, this command will restart the robot job and begin accepting new trajectories.

pbeeson, 2017-03-15:
@ted-miller This is correct.  To be clear, the issue here is that the motoman driver on the ROS side tries to establish connection and sends the START_TRAJ_MODE signal before sending EVERY command.  So, although an ESTOP being depressed will fail to move the robot, as soon as the ESTOP clears, the motoman driver  has no knowledge that its state is any different than startup (the CHECK_MOTION_READY just says "Failure/Unknown" Reulst/Subcode), so it just resestablishes the connection, set the START_TRAJ_MODE and starts moving the robot.  I guess the main issue is that the motoman driver only checks the CHECK_MOTION_READY status when the user is commanding, so it may not ever even see the estop was pressed and released in its current state.

I believe by adding something that runs the CHECK_MOTION_READY query at >= 10Hz, the driver should be able to detect an Estop and put itself into a state that requires the more explicit user override to continue.  This machinery would be akin to the enable/disable service calls I added to the industrial_robot_client last week but at the lower level of motoman_driver to be safer.  The other option over running the CHECK_MOTION_READY is to read the io of 80026, but that isn't generic enough for the indigo-devel branch.

pbeeson, 2017-03-15:
@ted-miller Alternatively, if you could use a different subcode for the CHECK_MOTION_READY Failure after an Estop has been pressed, then it would be easy to see that the Failure was from a recent estop, and force this user input before reconnecting and restarting.  As of right now, the Failure/Unknown Result/Subcode that happens when Motoplus ends the job on an estop is the same as when the motoman_driver first connects to the socket.

pbeeson, 2017-03-15:
Actually @ted-miller it turns out that the CHECK_MOTION_READY command does not inform the client that an estop is pressed.  The place the motoman_Driver knows this is when a START_TRAJ_MODE command fails and gets this info.  So there is no way on the PC side to "poll" the estop status other than to constantly send the START_TRAJ_MODE command.

Can you have the CHECK_MOTION_READY status be more informative over why it is in a failed state?

If not, then the easiest fix to this is simply to request the user to explicitly "Start/Enable" the robot each time CHECK_READY_STATUS fails, before sending START_TRAJ_MODE.

ted-miller, 2017-03-15:
IMO, the start_traj_mode signal should only be sent "as needed".  This could help control unexpected starting.

In regard to polling the check_motion_ready signal, I don't think it's needed.  If you connect to the state-server (which is the connection that reports joint positions), the estop-state is included in the status messages.

I agree with adding the subcode to check_motion_ready.  The subcode information is available for the trajectory commands, but check_motion_ready always returns zero.  I will update this shortly.

pbeeson, 2017-03-15:
OK.  For now I think the most afe thing to do is to require an outside signal to comes in anytime the CHECK_MOTION_READY command does not return Success.  Only then will the START_TRAJ_MODE signal be sent.

ted-miller, 2017-03-15:
I just added the subcode information to PR #150 .

gavanderhoorn, 2017-03-15:
@ted-miller wrote:
> IMO, the start_traj_mode signal should only be sent "as needed". This could help control unexpected starting.

I may be misunderstanding the code (or you), but I think that is already the case. See [MotomanJointTrajectoryStreamer::send_to_robot(..)](https://github.com/ros-industrial/motoman/blob/f79f76776aced9fbd636046c27d8415acac9726d/motoman_driver/src/joint_trajectory_streamer.cpp#L316-L322): 

```c++
bool MotomanJointTrajectoryStreamer::send_to_robot(const std::vector<SimpleMessage>& messages)
{
  if (!motion_ctrl_.controllerReady() && !motion_ctrl_.setTrajMode(true))
    ROS_ERROR_RETURN(false, "Failed to initialize MotoRos motion.  Trajectory ABORTED.  Correct issue and re-send trajectory.");

  return JointTrajectoryStreamer::send_to_robot(messages);
}
```

ted-miller, 2017-03-15:
> the issue here is that the motoman driver on the ROS side tries to establish connection and sends the START_TRAJ_MODE signal before sending EVERY command

@pbeeson  Can you clarify where in the code the START_TRAJ_MODE is being sent?

gavanderhoorn, 2017-03-15:
@ted-miller: that's in the lines I included in my comment above.

gavanderhoorn, 2017-03-15:
If you follow that you end up [here](https://github.com/ros-industrial/motoman/blob/58bc907dcfb038e5d2df351388b31e900e4f7999/motoman_driver/src/motion_ctrl.cpp#L58-L91), where both `controllerReady()` and `setTrajMode()` are implemented.

pbeeson, 2017-03-15:
@gavanderhoorn is right in that is "only" gets called when a client sends a new trajectory, but in an automated cell, the client may be blasting things constantly with no knowledge of the underlying estop state.  So as written, this is too liberal of a usage of the setTrajMode(true)

gavanderhoorn, 2017-03-15:
Would it be strange of me to think that the client should check the e-stop state by subscribing to the `RobotStatus` message?

I'm not saying that we couldn't add an additional layer in the driver, but ..

gavanderhoorn, 2017-03-15:
This would be separate from an explicit 'positive action' of the operator to resume normal operation though, if we interprete that strictly, I think.

pbeeson, 2017-03-15:
IMO (and the opinion of the Motoman team I'm working with in Michigan), for a production system, I don't think you should rely on your client to be "smarter" than the driver.  The lower you can push safety the better.  We were running a very simple test scripts to bang on motion and I/O, and those didn't have this.  move_to_joint.py doesn't have this.  You can't assume your users are going to be safe.  The driver needs to be safe.

gavanderhoorn, 2017-03-15:
As I wrote above, I'm not against putting this in the driver.

I do believe that it would be good behaviour of the client to check the status of the robot using the message that we've specifically created for these kinds of things.

pbeeson, 2017-03-15:
I agree that clients should do this (because they shouldn't just see failure and not know why), but at this point, after we saw this in the driver our customer is nervous about the safety of the driver.  I working on a fix at the driver level but we will also have safety from the status in our non-dumb clients.

pbeeson, 2017-03-15:
So,  I did create a simple solution to this, and tested, and it works.  Basically, connectionReady() is still checked before sending any motion, but only a service call (/motoman_enable) calls setTrajMode(true). (there is one for calling this with false if desired).  This is the functionality I wanted when I added similar to industrial_robot_client, but I didn't want to add it to every ROS-I robot driver out there, so I added it at the ROS-I level.  I think it's good to have at the lower-level as well at least for estops.  The current behavior forces the service call to happen anytime the connectionReady() is false, which includes startup, but am flexible on allowing it to autoconnect at startup of the motman_ros process if people want.

gavanderhoorn, 2017-03-15:
Sounds reasonable.

I think it would make sense if we can come up with some standardised naming here, so as to not make this completely Motoman-specific.

pbeeson, 2017-03-15:
Well there is already an /enable_robot in industrial_robot_client as per Issue #152, but this is more of a "Soft E-Stop" in that the low level robot driver is unaffected.  Since this is in the MotomanJointStreamer class I figured /motoman_enable was reasonable for naming.  This is similar to the other enable/disable except at the motoman level it sets the TRAJ_MODE to start or stop as needed.  But I'm open to any suggestions on fixing this better.  I need to move on to other Motoman driver issues now, but I'll push a PR for this later.  [stay tuned ;) ]

gavanderhoorn, 2017-03-15:
Nice.

It might make sense to actually look at some existing state machines for motion controllers and their interfaces. CANOpen comes to mind (CiA-402), or perhaps even PackML. I know there has been some discussion about introducing statemachines like that in *all* ROS-I components.

pbeeson, 2017-03-15:
PR #155

closed 2017-03-15 by pbeeson