ros-industrial/motoman #171: Document 'robot_enable' and 'robot_disable' services

gavanderhoorn, 2017-05-20:
#155 added the `robot_enable` and `robot_disable` services.

These services were never documented.

They should at least be mentioned in the [Using the Motoman FS/DX ROS Interface](http://wiki.ros.org/motoman_driver/Tutorials/Usage) tutorial and on the [wiki/Troubleshooting](http://wiki.ros.org/motoman_driver/Troubleshooting) page.

And possibly added to the [ROS-Industrial Robot Driver Specification](http://wiki.ros.org/Industrial/Industrial_Robot_Driver_Spec).

gavanderhoorn, 2017-05-20:
Related posting to the ROS-Industrial mailing list about this: [Important Motoman Driver Change (Indigo+)](https://groups.google.com/d/msg/swri-ros-pkg-dev/nYbNEwQOOtw/n5tjyYYuEwAJ).

shaun-edwards, 2017-05-22:
@nsnitish, can you take on?

nsnitish, 2017-06-01:
1. Added robot_enable and robot_disable services in [Robot_Controller](http://wiki.ros.org/motoman_driver/Tutorials/Usage#Robot_Controller). 
2. Added  [trajectory ABORTED](http://wiki.ros.org/motoman_driver/Troubleshooting#trajectory_ABORTED) error under ROS Errors in [wiki/troubleshooting](http://wiki.ros.org/motoman_driver/Troubleshooting).
3. Referred [Robot_Controller](http://wiki.ros.org/motoman_driver/Tutorials/Usage#Robot_Controller) in [ROS-Industrial Robot Driver Specification](http://wiki.ros.org/Industrial/Industrial_Robot_Driver_Spec)

@shaun-edwards! Can you please verify?

shaun-edwards, 2017-06-08:
@nsnitish, could you please provide links to the diffs, instead of the pages themselves.  It is difficult to know what actually changed.  For example, #1's diff is as follows: http://wiki.ros.org/action/info/motoman_driver/Tutorials/Usage?action=diff&rev2=14&rev1=10

nsnitish, 2017-06-08:
Following three pages have been updated:
1. [Robot_Controller](http://wiki.ros.org/action/info/motoman_driver/Tutorials/Usage?action=diff&rev2=14&rev1=10). 
2. [wiki/troubleshooting](http://wiki.ros.org/action/info/motoman_driver/Troubleshooting?action=diff&rev2=13&rev1=12).
3. [ROS-Industrial Robot Driver Specification for hydro+](http://wiki.ros.org/action/info/Industrial/Industrial_Robot_Driver_Spec/hydro?action=diff&rev2=9&rev1=7).

gavanderhoorn, 2018-05-01:
Cleaned up the `Usage` tutorial a bit to clarify the situation with the `robot_(en|dis)able` service.

This should be ok now.

closed 2018-05-01 by gavanderhoorn