ros-industrial/motoman #172: IKfast solution for MH5 Robot

shaun-edwards, 2017-05-22:
Please create an IKFast solution for the [MH5](https://github.com/ros-industrial/motoman/tree/kinetic-devel/motoman_mh5_support).  To achieve this, perform the following:
 - [ ] Create an IKFast solution (see [tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/ikfast_tutorial.html)).  Note, there is a [docker](https://hub.docker.com/r/personalrobotics/ros-openrave/) of openrave that is known to work.
 - [ ] Create a test program that verifies the resulting solution returns the nearest IK solution in joint space.

shaun-edwards, 2017-05-22:
@nsnitish, can you address this?

@gavanderhoorn, are you aware of anything that verifies the solution (i.e. the second item)?  Based on my experience, this is not the case.  Perhaps such a test should be added to the [kinematics_test](https://github.com/ros-planning/moveit_kinematics_tests) repo.

gavanderhoorn, 2017-05-22:
> @gavanderhoorn, are you aware of anything that verifies the solution (i.e. the second item)?

yes, OR actually has built-in support for testing these plugins. but that is not used with the command line that the tutorial gives you (which is actually not the correct one).

See [How to verify generated IKfast plugin?](http://answers.ros.org/question/227219) on ROS Answers.

gavanderhoorn, 2017-05-22:
Btw, before such an IKFast plugin is generated, I would first address #122.

shaun-edwards, 2017-06-01:
FYI...#122 has been addressed.

nsnitish, 2017-06-05:
@gavanderhoorn , In [motman_mh5 catalog](http://www.motoman.co.uk/uploads/tx_catalogrobot/MH5_E.pdf) it is mentioned that it is a 6 axis robot. Also, from openrave's robot info I can see 6 links in addition to the base link. Can you please confirm if it is a 6 or 5 DOF robot?
Also, ikfastdemo.cpp referred in [How to verify generated IKfast plugin?](http://answers.ros.org/question/227219/how-to-verify-generated-ikfast-plugin/) is not available. Do you know of any other source to verify the ikfast solution?

gavanderhoorn, 2017-06-05:
> In [motman_mh5 catalog](http://www.motoman.co.uk/uploads/tx_catalogrobot/MH5_E.pdf) it is mentioned that it is a 6 axis robot. Also, from openrave's robot info I can see 6 links in addition to the base link. Can you please confirm if it is a 6 or 5 DOF robot?

The MH5 is a six axes robot.

> Also, ikfastdemo.cpp referred in [How to verify generated IKfast plugin?](http://answers.ros.org/question/227219/how-to-verify-generated-ikfast-plugin/) is not available. Do you know of any other source to verify the ikfast solution?

I don't use `ikfastdemo.cpp` in my answer to the question you linked.

OpenRAVE will run the tests for you, there is no need to code anything. Have you read the documentation that I link in that question? Especially [OpenRAVE Documentation - inversekinematics Module - Testing](http://openrave.org/docs/0.8.0/openravepy/databases.inversekinematics/#testing)?

shaun-edwards, 2017-06-05:
@nsnitish, you may want to pull down the latest version of the mh5...see fixes in PR #175

nsnitish, 2017-06-05:
@gavanderhoorn ! Thank you for confirming.
@shaun-edwards ! Thanks for the update.

shaun-edwards, 2017-07-04:
@nsnitish, almost there on this one.  Can you please add a PR for just the ikfast solution to the `motoman_experimental` [repo](https://github.com/ros-industrial/motoman_experimental)

gavanderhoorn, 2018-05-01:
I believe this has been completed:

 - `motoman_experimental` contains the MH5 IKFast plugin
 - the testing is done somewhere else / done using OpenRave

closed 2018-05-01 by gavanderhoorn