ros-industrial/motoman #174: Joint feedback - Velocity and Acceleration

ted-miller, 2017-05-23:
It just came to my attention that the Joint Feedback messages have fields for velocity and acceleration.  It would appear that this data gets populated into the velocity/effort fields of [rostopic echo /joint_states].

However, the MotoROS driver doesn't actually populate these fields.  The [validFields] mask is set to 2 for position only.

Is this ok?  Should I be populating the velocity/acc fields?  Is this data useful for anything?

gavanderhoorn, 2017-05-23:
If the controller makes that data available, then yes please, populate it. Same goes for the `effort` fields (`Ros_CtrlGroup_GetTorque(..)`?).

As to the usefulness: I would say it's always useful to have as much state information available as possible, even if right now use-cases might not be apparent.

Personally I would however not derive those values (by dx/dt fi).

ted-miller, 2017-05-23:
I can get the feedback-speed and torque for each axis on FS100, DX200, and YRC1000 (not DX100).

However, as best I can tell from looking at MotoROS and the Industrial Core, the joint feedback message isn't designed to accept effort/torque.  Are you suggesting that I overwrite the acceleration field with the torque?  Or that I add a new member to the feedback-message?  If I increase the size of the feedback message, will that break anything in the Industrial Core? (I haven't delved into it in great detail yet)

shaun-edwards, 2017-05-23:
The ROS side and MotoPlus size message definitions have to match.  We could make that happen, but I'm more worried about sending even more data through the controller.  Does the controller have a max packet size for socket comms?

ted-miller, 2017-05-23:
Probably...  but I expect that the TCP stack on the controller should accommodate the normal MTU for ethernet, which is 1500 bytes (need to test, of course).  If I add a torque field, it would put a SmBodyJointFeedbackEx packet at 692 bytes.

gavanderhoorn, 2017-05-24:
@ted-miller wrote:
> However, as best I can tell from looking at MotoROS and the Industrial Core, the joint feedback message isn't designed to accept effort/torque.

O right, I misremembered `JOINT_FEEDBACK_EX` having the `effort` fields as well. It is only available in IRCv2.

> Are you suggesting that I overwrite the acceleration field with the torque? Or that I add a new member to the feedback-message? If I increase the size of the feedback message, will that break anything in the Industrial Core? (I haven't delved into it in great detail yet)

Well, we could, but that would mean an update to the ROS side as well. Not sure that is worth it.

gavanderhoorn, 2017-05-24:
@shaun-edwards wrote:
>  I'm more worried about sending even more data through the controller. Does the controller have a max packet size for socket comms?

I'm not too worried about this. The TCP/IP stack on the Motoman controllers is pretty sane (after all, it's vxworks), and MotoPlus applications have direct access to that.

gavanderhoorn, 2017-11-23:
@ted-miller: would adding the velocity data to the `JOINT_FEEDBACK_EX` msgs be something you could implement?

For some things we're starting to look at here it might come in handy to have that data available.

ted-miller, 2017-11-27:
Sure.

gavanderhoorn, 2017-11-27:
And a question on ROS Answers about this: [Motoman FS100: Missing velocity and effort in joint_states](https://answers.ros.org/question/276723).

(note that `velocity` and `effort` aren't missing, I'm pretty sure the OP misremembered).

ted-miller, 2017-11-27:
@gavanderhoorn  See https://github.com/ted-miller/velocity_feedback 
I haven't actually tested this yet beyond compiling.

gavanderhoorn, 2017-11-27:
Thanks.

Did you remove all pkgs except `motoman_driver` in that commit as well?

For the future; could you please commit such large changes in a separate commit? The additions for the velocity feedback are now burried in a commit that touches 351 files ..

ted-miller, 2017-11-27:
Yeah... I realized that after I committed.  Let me know if you need me to redo this repo.

I've attached a comparison report that shows the files that were actually changed.

[image]

gavanderhoorn, 2017-11-27:
It would be really nice if you could push the same changes, but then either not delete the pkgs, or do everything in two commits.

ted-miller, 2017-11-27:
I just reverted and re-pushed the changes.

gavanderhoorn, 2017-12-01:
Just tested this.

Haven't validated the output (ie: correspondence between commanded and reported velocities), but MotoROS seems to function as it should and I get sane data on the `/joint_states` topics.

Is there a rate limit on the velocity feedback that MotoPlus makes available? It appears to lag a bit and seems a bit discretised as well.

[image]

ted-miller, 2017-12-04:
I'm not sure. I've never actually used this API before, but I expected it to have the same behavior as the feedback-position API.

I'll need to run some additional tests.

ted-miller, 2017-12-06:
I have run some additional testing and got similar results to yours.  I compared the values to the [Servo Monitor] function on the pendant and they are identical (including the aggregation/discretization).  So, I believe that this result is the best I can obtain using the MotoPlus API for feedback speed.  I will prepare a report and send to HQ in Japan... but this will take a long a time to discuss and potentially improve.

There is one other way to obtain the feedback speed without additional hardware.  I ran a quick test and found that this does provide data with finer granularity. But there are couple drawbacks:

- I suspect the delay for the feedback will be greater because the data has to get routed through a few layers in the controller before I can access it in MotoROS.  But, given that it's a more accurate representation of the instantaneous speed, maybe this is a moot point. I'm going to run additional testing today (or maybe later this week) to quantify the timing.
- There are a bunch of protected robot parameters which must be enabled to use this function.  Since we just got away from requiring yaskawa-level parameters, I don't want to go back to it.  I have an idea of how it could possibly be automated in MotoROS... but that piece could not be published open-source.  I would need to put it in a pre-compiled library.
- This would consume a LOT of robot M registers.  If someone starts to combine MotoROS with other robot options, it's very possible there could be a resource conflict. 

While I'm at it, is this a good time to implement feedback torque?  If there were some kind of version-check message implemented in motoman-driver, I could use it to maintain backward compatibility of the MotoROS portion.

gavanderhoorn, 2017-12-06:
Thanks for the update.

> I suspect the delay for the feedback will be greater because the data has to get routed through a few layers in the controller before I can access it in MotoROS. But, given that it's a more accurate representation of the instantaneous speed, maybe this is a moot point.

Delay is undesirable. But discretisation is also undesirable. I can't really decide whether one is worse than the other. In most cases delayed data is unusable, whether it's discretised or not (too old == too old). And for applications where info about velocity is interesting, I believe that delay would always be undesirable.

> There are a bunch of protected robot parameters which must be enabled to use this function. Since we just got away from requiring yaskawa-level parameters, I don't want to go back to it. I have an [...]

That would not be different from how the 'parameter extraction' works right now, correct?

> This would consume a LOT of robot M registers. If someone starts to combine MotoROS with other robot options, it's very possible there could be a resource conflict.

That could definitely be a concern. Personally I think it's nice to have velocity feedback available, but not at all costs.

> While I'm at it, is this a good time to implement feedback torque? If there were some kind of version-check message implemented in motoman-driver, I could use it to maintain backward compatibility of the MotoROS portion.

There is no versioning in the protocol unfortunately, so that is not something we can exploit.

The `MSG_MOTO_JOINT_FEEDBACK_EX` does not include effort, only acceleration. If we want to add this, we'd either have to repurpose the acceleration field or introduce a new message (IRCv2 does actually include `effort` in its 'Dynamic Joint State' message).

ted-miller, 2017-12-06:
I have performed testing to compare the MotoPlus API vs the M-Register method to read feedback speed.  It turns out I was wrong about the delay.  The M-Register method is definitely superior.
[FeedbackTester Results.xlsx](https://github.com/ros-industrial/motoman/files/1537062/FeedbackTester.Results.xlsx)
So, I need to investigate the viability of deploying the M-Register method and automating it inside of the param-extraction library.  (I'm not even sure if this is an "official" controller feature.)

gavanderhoorn, 2017-12-07:
That's certainly an improvement @ted-miller:

[image]

(ignore the sign on the second part of the plot, it's incorrect).

Both the delay as well as the discretisation of the MotoPlus API feedback is clearly visible here.

ted-miller, 2017-12-28:
@gavanderhoorn See c1cec328cc2caae792301ca6062b78cc7cb617ef
I have implemented the change to the Param Extraction library to automatically enable all the parameters for high-speed Register Speed Feedback. It turns out that these are all user-level parameters (not protected)... but there's a bunch of them and this method helps to eliminate confusion/error.
At startup, the library will attempt to read the param values.  If they are not set, it will automatically enable them.  This will require a reboot the first time it is done, so I raise an alarm, "8001 [10] Speed FB enabled, reboot now". After that, velocity data will be included with the Joint Feedback messages.

The high-speed register method is only available for DX200 and newer.  So, for DX100 and FS100, I am disabling velocity feedback entirely.  I figure that MotoPlus-API method (above) is pretty useless.  Thoughts?

If this gets merged into the main branch, I plan to add some wiki pages with these details.

ted-miller, 2018-01-09:
Are there any known conflicts or issues if I merge these changes into the main branch and submit a PR? I'm assuming that the existing ROS-I code can handle velocity data being transmitted, even if it's not used by the final application.

gavanderhoorn, 2018-01-09:
I would suggest creating a new branch on your regular `motoman_driver` fork. Commit the changes to that fork, and open a PR from that fork against `ros-industrial/motoman`.

That way even if a review / testing round takes some more time, you won't be 'blocked', as you can just work in a different branch.

I can't test this - as I only have an FS100 - but perhaps we can find someone with a DX200 or YRC1000 that could take a quick look.

I'll take a look at the ROS side to make sure that the deserialisation results in proper `JointState` msgs with the velocity fields populated. I'm assuming it's all ok, as it already worked with what you posted in https://github.com/ros-industrial/motoman/issues/174#issuecomment-347301240, but it's good to check in any case.

ted-miller, 2018-01-10:
I'll ask Erc Marcil to validate on my end.

ted-miller, 2018-04-11:
@gavanderhoorn 
If you need this feature on the FS100, you could try the API mpSvsGetVelTrqFb. This gets speed and torque feedback at a good rate. (I only recently learned of this API.)

gavanderhoorn, 2018-04-11:
Thanks. That is good to know.

Would you accept a PR adding that to the `FS100` code paths in `motoros`? :)

ted-miller, 2018-04-11:
Yeah.  I can probably get around to it on Monday.  Or, feel free to submit it sooner.

ted-miller, 2022-04-22:
@nexus1203, The "register method" is a feature of the robot controller. When enabled, the robot controller will automatically populate M-Registers with the percentage of max torque. (Not an absolute Nm.) You would then need to move this data out of the robot using High Speed Ethernet Server, or MotoPlus, or a fieldbus connection like Ethernet/IP.

If you are using ROS, then there is a function in the code to read the feedback torque (Nm). https://github.com/ros-industrial/motoman/blob/7860ff545106d98ed6a3fa0121f2fb60891f69bd/motoman_driver/MotoPlus/CtrlGroup.c#L454-L477

But, this function isn't actually called by anything, nor is the data linked to any feedback messages from MotoROS > motoman_driver. So, you could add this functionality with some modifications to both MotoROS and motoman_driver.

Steps on the MotoROS side:
•	Add a new [command ID](https://nam10.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fros-industrial%2Fmotoman%2Fblob%2F642dad8f955a6003be6774ca994a1c56a67a7d09%2Fmotoman_driver%2FMotoPlus%2FSimpleMessage.h%23L52-L81&data=04%7C01%7Cted.miller%40motoman.com%7C22511cfa056f4cea9c4208da1d678db9%7C5bff86433f854a0a80f929eae318b5ce%7C0%7C1%7C637854627855240838%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=PB6eDabBwcFL2RgNKCvxyo8L3pmXhNHa8yrTVUHABmA%3D&reserved=0) and [message format](https://nam10.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fros-industrial%2Fmotoman%2Fblob%2F642dad8f955a6003be6774ca994a1c56a67a7d09%2Fmotoman_driver%2FMotoPlus%2FSimpleMessage.h%23L392-L416&data=04%7C01%7Cted.miller%40motoman.com%7C22511cfa056f4cea9c4208da1d678db9%7C5bff86433f854a0a80f929eae318b5ce%7C0%7C1%7C637854627855397346%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=%2BsJI9HRT7ayuakrLdObuDy1VMTLNc65%2F4gWOKCoC%2FTk%3D&reserved=0) for SimpleMessage, which is the communication between robot controller and PC. 
•	Add code to the StateServer to generate this feedback message and send it to the PC. I recommend adding this in the [same spot that sends position feedback](https://nam10.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fros-industrial%2Fmotoman%2Fblob%2F642dad8f955a6003be6774ca994a1c56a67a7d09%2Fmotoman_driver%2FMotoPlus%2FStateServer.c%23L132&data=04%7C01%7Cted.miller%40motoman.com%7C22511cfa056f4cea9c4208da1d678db9%7C5bff86433f854a0a80f929eae318b5ce%7C0%7C1%7C637854627855397346%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=cOdmk1VFwJ745orSN1cHih284%2BmzhCfMKmUr1XU1F3U%3D&reserved=0).

Alternatively, you could modify the format of the [existing feedback message](https://github.com/ros-industrial/motoman/blob/642dad8f955a6003be6774ca994a1c56a67a7d09/motoman_driver/MotoPlus/SimpleMessage.h#L211-L220) to include the torque data. 

Either option require modifications to the motoman_driver code on the PC. It needs to either accept the newly created command, or accept the modified format of the existing feedback command.

nexus1203, 2022-04-23:
@ted-miller Thank you very much for your explanation.
May you please point me to a documentation for this method? I went through the manuals of "high speed Ethernet server" and "Ethernet IP server" functions but I can't seem to find a reference to how to enable this function. I do know I can read M variables using these variables but I am trying to figure out how to actually move data to M variable.

ted-miller, 2022-04-25:
I don't believe that any formal documentation exists for this feature. I think it is technically not an "officially supported" feature. But, I did just verify that it works on my YRC1000.

S1CxG122: Enable/Disable. Each bit corresponds with an axis. 

    - Ex: Bit 0 = [Enable S Axis], Bit 1 = [Enable L Axis], etc...
	
DX200: S1CxG488 ~ 495: Register number for each axis' value
YRC1000: S1CxG910 ~ 919: Register number for each axis' value

    - Ex: [S1CxG 488/910 = 50]: M050 will contain the torque percentage for S axis, if [S1CxG122:Bit0 == 1]. The value is a signed value, based on the direction of rotation.

nexus1203, 2022-05-05:
@ted-miller  After trying for over a week and so. I am still not able to read M-registers on DX200. I am using high-speed ethernet function and generally M register should be available at 0x79 (Register data reading / writing command). However when reading these registers, the value is always  returned as 0. In your previous comment you mentioned:
> When enabled, the robot controller will automatically populate M-Registers with the percentage of max torque
Is it function needs to be enabled manually? On the controller side, high-speed ethernet function is enabled, all the other commands such as 0x76 and 0x77 works fine. But I am really hitting a wall with M register method. 
For the context, What I am trying to achieve here is a reliable way to measure torque reading (preferably >= 200Hz). However, I am stuck with 25ms update rate on ROS, and 10ms request/response time on high speed ethernet function. From my understanding High speed ethernet function has fixed request/response time and any faster than 10ms communication is not possible (I am really hoping that I am wrong).

ted-miller, 2022-05-05:
Yes, this function needs to enabled manually, using the robot parameters listed above. 
- Put the pendant into TEACH mode
- Upgrade to MANAGEMENT security level on the teach pendant.
  - Touch `System Info` > `Security`.
  - Press SELECT button to open the drop-down. Choose MANAGEMENT and press SELECT.
  - The default password is `99999999`. Press ENTER.
- From the main menu, touch `Parameter` > `S1CxG`
  - Set `S1C1G122` = 63  //First six bits 
  - Set `S1C1G488` = 50  //M050 for S axis
  - Set `S1C1G489` = 51  //M051 for L axis
  - Set `S1C1G490` = 52  //M052 for U axis
  - Set `S1C1G491` = 53  //M053 for R axis
  - Set `S1C1G492` = 54  //M054 for B axis
  - Set `S1C1G493` = 55  //M055 for T axis

Then you can validate the function on your teach pendant by touching `In/Out` > `Register`. Navigate to M050 - M055 and watch these values as you jog the robot arm.

If the high-speed-ethernet commands are still returning zero, then please use Wireshark to record the network traffic between your PC and the robot controller. 

> From my understanding High speed ethernet function has fixed request/response time and any faster than 10ms communication is not possible (I am really hoping that I am wrong).

It is not actually a fixed time. But, it is a relatively slow protocol. I expect a single command to read an M-Register would take between 5-10 ms. But you need to send 6 commands to read all the axes.

There are two ways to achieve 200Hz (or greater). Both of these would require paid options for the robot.
1. Use the MotoPlus SDK. You could either modify MotoROS to speed up the feedback publisher. Or, you could create your own custom application specifically for reading torque.
2. Using the M-Register method above, you could modify the robot's CIO Ladder Program to automatically move the Register data into External Outputs. (This is updated at a fixed 2 ms interval.) From there, you could utilize some fieldbus interface (such as Ethernet/IP or Profinet) to send the data to external device. I know there are open-source implementations of Ethernet/IP available on GitHub. You would have to integrate that with you application to read the data in real time.

nexus1203, 2022-05-05:
@ted-miller Thank you very much for the setup tutorial. I will test this the first this tomorrow.
> There are two ways to achieve 200Hz (or greater). Both of these would require paid options for the robot.

I will contact my local dealer to get quotations on these. Again, thank you for your guidance.

gavanderhoorn, 2022-05-05:
@nexus1203 (and @ted-miller also ;) ): could you please start a new discussion if, after testing, it still doesn't work for you?

Your questions are off-topic for this issue.

We should try to keep issues on-topic, and using HSES to access M-registers is rather different from what was being discussed in this issue originally.