ros-industrial/motoman #190: Two [ERROR] when use with SDA5F(FS100 v3.24)

tylr2005, 2017-12-27:
roslaunch motorman_driver robot_interface_streaming_fs100.launch robot_ip:=10.0.0.2
[ERROR]Failed to find topic_list parameter
[ERROR]Message callback for message type: 2017, not executed

shaun-edwards, 2017-12-27:
Please provide the complete log output.  What moveit config/URDF are you using?

tylr2005, 2017-12-27:
The two errors after involving “roslaunch motoman_driver robot_interface_streaming_fs100.launch robot_ip:=10.0.0.2”.  Don’t need involve moveit.

Then I download https://github.com/qqfly/motoman_sda5f_pkg and use “roslaunch motoman_sda5f_support  robot_interface_streaming_sda5f.launch”, it works ok.

gavanderhoorn, 2018-01-24:
@tylr2005: I'm closing this, as your last comment seems to indicate that it's no longer a problem for you.

The error message you quote points to a mis- or unconfigured `topic_list` parameter.

Starting the `robot_interface_streaming_fs100.launch` in isolation, without any additional parameters set (such as `topic_list`) is not going to work. The `robot_interface_streaming_sda5f.launch` most likely sets exactly those parameters that `motoman_driver` needs to function properly.

closed 2018-01-24 by gavanderhoorn