ros-industrial/motoman #193: MotoROS: add a (simple) ui on the TP

gavanderhoorn, 2018-01-24:
A discussion topic / enhancement request: the MotoROS side of the driver could include a UI shown on the TP that provides both access to settings as well as status information.

Some possible pieces of information / configuration options such a UI could provide (from https://github.com/ros-industrial/motoman/pull/191#issuecomment-359940450):

 - (en|dis)abling velocity feedback (recent addition in #191)
 - maybe enable/disable torque feedback
 - display a diagnostic monitor screen
 - display all the controller information that is calculated at startup (max speeds, pulse->radian, etc)

gavanderhoorn, 2018-01-24:
Some more suggestions:

 1. MotoROS version info
 1. connection status: connected, disconnected, remote IP, etc
 1. how many connections to the State, Motion and IO servers are being serviced (and how many are still available), perhaps also nr of tasks running
 1. Motion groups detected vs controlled (could be part of `4` in the OP)
 1. message statistics:
    1. corrupt msgs (if ever)
    1. unknown msgs
    1. unhandled msgs (ie: IO msgs to state server)
    1. total nr of msgs (nr of trajectory msgs, etc)
 1. trajectory execution status: nr of traj msgs received since start of new trajectory, etc. We don't know how long trajectories are (they are streamed), so this would essentially be a running counter that gets reset upon reception of a new traj)
 1. `Ros_MotionServer_IncMoveLoopStart` queue status
 1. simple 'log viewer' (telnet output into multi-line text edit control)
 1. current joint states per group as MotoROS 'sees' them (to diagnose: start state deviated from current state problems)

gavanderhoorn, 2018-01-26:
Things to potentially configure:

 - update/control rates (instead of recompiling the MotoPlus binary, change rate and restart controller / reconnect)
 - switch between motoman and ROS joint ordering (optional, might not have much use)
 - (en|dis)able 'single client mode': only allow a single IP to setup state and motion control connections (the latter is already impossible, but just to be thorough)
 - (en|dis)abling motion control side of MotoROS only: state connections keep working, but external control of the robot is impossible
 - (en|dis)abling IO interface of MotoROS (in both the motion and io servers)
 - (en|dis)abling MotoROS completely, without having to uninstall the binary
 - (en|dis)abling motion groups reported on and controlled by MotoROS (robot has 3 groups, I only want to allow control of and report state of one of those)

gavanderhoorn, 2018-01-26:
@ted-miller: I just wrote down what I could think of. In no particular order.

Would there be anything that you feel should be looked at first?

ted-miller, 2018-02-01:
I'll make some dummy screens sometime next week and post them to see what you think. These are the items I think should be addressed first, as they require less dramatic changes to the MotoPlus:
 - (en|dis)abling velocity feedback
- display all the controller information that is calculated at startup (max speeds, pulse->radian, etc)
 - MotoROS version info
 - connection status: connected, disconnected, remote IP, etc
 - how many connections to the State, Motion and IO servers are being serviced (and how many are still available), perhaps also nr of tasks running
 - Motion groups detected vs controlled (could be part of 4 in the OP)
 - Ros_MotionServer_IncMoveLoopStart queue status
 - current joint states per group as MotoROS 'sees' them (ie: ROS joint order, to diagnose: start state deviated from current state problems)
 - update/control rates (instead of recompiling the MotoPlus binary, change rate and restart controller / reconnect)

ted-miller, 2018-02-01:
Do you guys have localized versions of ROS Industrial utilities in languages other than english?  I'm wondering if these screens will ever be multi-lingual.

Given the limited screen space on the pendant, I need to plan extra carefully if I know the app will display other languages.  (German and Mandarin are especially challenging.)

gavanderhoorn, 2018-02-02:
> Do you guys have localized versions of ROS Industrial utilities in languages other than english?

No, not at this time.

Focusing on English seems ok to me, but perhaps @shaun-edwards has some other input.

> German and Mandarin are especially challenging

I can imagine. Are you doing those yourself as well?

gavanderhoorn, 2018-03-02:
Curious/excited: :bellhop_bell: ? :)

ted-miller, 2018-03-02:
Sorry, I haven't had much time to work on side-projects lately.  Here's what I have so far.  I'm still not sure the best way to display all the information listed above.  Your feedback is welcomed.

[image]

gavanderhoorn, 2018-03-03:
Looks ok. I'm guessing this is a YRC1000 TP in the screenshot?

Aesthetically I'm not such a fan of centre alignment, perhaps all left aligned would be cleaner?

>  I'm still not sure the best way to display all the information listed above. Your feedback is welcomed.

Is the list the 'only' list we have? Or are the other tabs/buttons also available?

gavanderhoorn, 2018-03-03:
Btw: no connection to the IO server isn't necessarily a problem, right? I guess the exclamation mark being in a green square means it's just a warning, but it looks like something should be done to correct it.

ted-miller, 2018-03-03:
This is how it will look on any controller model. It's a custom Windows CE application that runs overtop the standard interface.  In regards to what controls are available... I can do anything. Just think of it as a normal Windows app. 

No, the I/O server is not required.  I can certainly play with the icons/colors.  I simply had the exclamation icon already made from another application.

gavanderhoorn, 2018-08-13:
Just curious: have you had any more time to work on this @ted-miller? I'm liking the panel you've posted. It does seem like something that would be very useful for MotoROS users (not just ROS users).

ted-miller, 2018-08-13:
No, I haven't touched this since March. It's still on my to-do list, but it's at the bottom of that list.

Please keep this issue open.

gavanderhoorn, 2019-09-19:
@ted-miller: have you had any time to work on this? The mock-up you showed would make a very nice addition to MotoROS.

Right now status of the connection and other properties of the driver and ROS connection are invisible on the controller. Having a display like you showed in https://github.com/ros-industrial/motoman/issues/193#issuecomment-369964241 would be invaluable.

ted-miller, 2019-09-20:
No, I haven't touched this. Honestly, the chances are low that it'll happen anytime soon.

However, if you want easy visibility of those signals, I could create a quick/easy I/F panel. Here's an example:
[image]

For north/south America, the I/F Panel function should be enabled on every arm. I don't know if this function is enabled by default in other regions. Can you check your controller to see if this button exists on the bottom of your pendant screen?

gavanderhoorn, 2019-09-20:
> Honestly, the chances are low that it'll happen anytime soon.

Could someone else do this work? What would be needed?

> However, if you want easy visibility of those signals, I could create a quick/easy I/F panel. Here's an example:
>
> [..]
>
> For north/south America, the I/F Panel function should be enabled on every [...]

Yes, I have the I/F Panel button.

ted-miller, 2019-09-23:
> Could someone else do this work? What would be needed?

I'll check with Eric to see what he has going on.
This project requires our Pendant SDK. Additionally, it requires a very specific version of Visual Studio that supports Windows CE.

gavanderhoorn, 2020-06-26:
@ted-miller: would there be any way to (re)start this effort?

ted-miller, 2020-06-29:
@EricMarcil: Is this something you could work on? I've got a skeleton app, but I basically abandoned it due to higher priorities. If you're also busy, I can check if someone else in the department is available.

ted-miller, 2020-06-29:
@gavanderhoorn : I'll still offer up the I/F panel option. It far less aesthetically appealing. But, I could have that ready for you today.

EricMarcil, 2020-06-29:
@ted-miller: I'm pretty busy recently but send me what you have, I'll try to look at it when I have a chance.

gavanderhoorn, 2020-06-30:
@ted-miller wrote:
> @gavanderhoorn : I'll still offer up the I/F panel option. It far less aesthetically appealing. But, I could have that ready for you today.

I don't have an immediate need myself.

But there have been a few questions about status and other runtime aspects of MotoROS which I felt could be captured/displayed by the UI being discussed here in this issue.

That's why I thought to ask whether we could restart this effort.

gavanderhoorn, 2020-10-07:
Just making sure again this doesn't fall off everyone's radar again. :bell: :robot:

ted-miller, 2020-10-07:
Realistically, the custom TP GUI is way off my radar. Though, I'm still willing to whip up an I/F Panel. It's better than nothing...

gavanderhoorn, 2025-03-19:
I would still *really* like to add a GUI for MotoROS to the TP, but I'm assuming that's not going to happen anymore for MotoROS1.

The same issue tracking this for MotoROS2: https://github.com/Yaskawa-Global/motoros2/issues/40 (just so we keep things connected).

closed 2025-03-19 by gavanderhoorn