ros-industrial/motoman #195: wiki: link to MotoROS binaries in repository instead of attaching them

gavanderhoorn, 2018-01-26:
Suggestion: let's link to the MotoROS binaries hosted here on Github from the wiki pages instead of attaching them.

By linking directly we can avoid having to keep the pages up-to-date, as users will automatically download the latest releases.

gavanderhoorn, 2018-01-26:
If we're using the attachments to do controlled releases of the binaries we can/should of course keep doing that, but I'm not aware of such a policy.

ted-miller, 2018-02-01:
I used attachments for two reasons.
- Many new users are not familiar with Github.  The web interface is weird and non-intuitive for downloading a single file. The wiki attachment is easier.
- If the default branch changes (kinetic-devel), then we'll have to update the wiki links anyway.

Having said that, it's not a big deal.  I don't care that much either way.

gavanderhoorn, 2018-02-02:
> Many new users are not familiar with Github. The web interface is weird and non-intuitive for downloading a single file. The wiki attachment is easier.

we can directly link to the binaries like so: https://github.com/ros-industrial/motoman/blob/4b9fd56dbff9d6aeabb4c6614b0723e6a2ee060c/motoman_driver/MotoPlus/output/DX100/MotoRosDX100/MotoRosDX100.out?raw=true

> If the default branch changes (kinetic-devel), then we'll have to update the wiki links anyway.

that is true, but I expect that to happen a lot less than releasing new versions of MotoROS.

gavanderhoorn, 2018-02-02:
We could also make use of Githubs [releases](https://github.com/ros-industrial/motoman/releases) system. But that would require some manual work ([docs](https://help.github.com/articles/creating-releases/)).

ted-miller, 2018-04-17:
Yeah. I'll fix the links.  I just forgot about this.

ted-miller, 2018-04-17:
It is done.

closed 2018-04-17 by ted-miller