ros-industrial/motoman #196: MotoROS: embed version in filename?

gavanderhoorn, 2018-02-02:
The MotoROS binaries all have the same name: `MotoROSccccc.out` (where `ccccc` is the controller series).

It would perhaps be nice to somehow add the version of the binary in there.

Personally I've always renamed the files (after downloading) to: `motoros170.out` (for `v1.7.0`).

I'm not sure how much manual work this would be (or whether it could be automated in the VS build scripts), but it could be a nice enhancement.

gavanderhoorn, 2018-05-03:
@ted-miller: is this something you would consider?

One thing I did not take into account is that this complicates linking from the wiki (#195).

ted-miller, 2018-05-03:
I'm fine with this idea.  There have been some questions lately about version; I think this would help.

My only concern is keeping the filename less than 16 characters.  (Perhaps you remember from a while back the problem with loading a .out file with an "invalid" file name.)

So, here's what I'm thinking.  Is it too cryptic?
mROS_DX1_v180
mROS_DX2_v180
mROS_FS1_v180
mROS_YRC1_v180
mROS_YRC1u_v180

gavanderhoorn, 2018-05-03:
I've used `motorosXYZ.out`, where `XYZ` is `173` (for 1.7.3) for instance.

ted-miller, 2018-05-03:
I want to keep the controller model in the filename.  The controller gives a very cryptic alarm if you load a .out file that is compiled for the wrong controller.

gavanderhoorn, 2018-05-03:
O right, I had not considered that.

What about `ABCmotorosXYZ.out`. That is 13 chars for the filename, 3 for the extension.

`ABC` == controller, `XYZ` == version.

I seem to remember I had run into some problems with underscores in filenames, but it could be that I'm misremembering things.

gavanderhoorn, 2018-06-19:
#213 added this.

@ted-miller: we'll need to remember updating the links on the wiki on the [installation tutorial](http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer).

closed 2018-06-19 by gavanderhoorn