ros-industrial/motoman #214: Automatic speed limiting in MotoROS?

ted-miller, 2018-05-15:
Someone proposed today that we modify the MotoROS code to automatically limit the incremental data if it exceeds the maximum speed-limit for any axis.

I have a few concerns.  I'm interested in hearing other people's thoughts.

1. If one axis gets limited, we need to scale the other axes proportionally. Although this could probably be implemented on a single-arm system, it becomes more complicated on a multi-arm system. I don't think we have any mechanism to know whether the other control-groups are supposed to be synchronized or independent.

1. Ignoring the multi-group scenario above, is there consequence to the robot not behaving the commanded trajectory?  The truncated position would have be added on during the next time cycle... so is it a problem in the ROS environment if the robot's speed fluctuates around the commanded speed?

1. From a usability perspective, is it best to stop the robot and alarm (notify the user that they're trying to go too fast so that they can correct the process)?  Or is it better to keep running at the expense of expected performance?  I can think of scenarios where either one could be better than the other.

Anyone have any thoughts?

gavanderhoorn, 2018-05-15:
Are you specifically referring to *scaling down* increments when they would lead to resulting motions going outside robot limits?

ted-miller, 2018-05-15:
Yes

gavanderhoorn, 2018-05-15:
I would not be in favour of that.

The behaviour I've observed in these sort of situations (our SIA20F on a rail) is that the safety systems of the controller will abort any motion as soon as it violates a limit, and the TP will show an error.

This seems to me the 'ideal' situation:

 1. it's immediately clear something is wrong
 1. it's already implemented
 1. it aligns with how this would work with normal/INFORM motions (afaik, if it were even possible to command motions outside the limits)
 1. it's transparent (scaling would not be)
 1. it's intuitive
 1. has none of the corner cases you mention
 1. does not require special handling on the ROS side

And finally: it's *explicit*. I cannot overstate how important that is. Plodding along at scaled down speeds just so the robot will at least do *something* is not a good thing. Being explicit about what is and isn't possible and supported *is*. There are too many places in ROS (and the pipeline from planning, motion generation, execution, driver and controller) already that leave things implicit.

Could you perhaps give a bit more of the rationale for wanting to do this? I'm pretty sure I'm not seeing something scaling would make easier/better/desirable.

> So is it a problem in the ROS environment if the robot's speed fluctuates around the commanded speed?

Well .. yes. At least for MoveIt's trajectory execution monitor. That part of MoveIt monitors the execution of trajectories and will step in if things take too long or are stalled - typically by aborting execution completely, but that is configurable / extendable. Note that this is not a safety-system, but a first check in the pipeline to not unnecessarily wait for completion of actions that are stalled fi.

If the entity that executes a trajectory (in this case the controller, but could be something else) starts changing the way it executes trajectories without the monitor being aware of this, it'll most likely conclude that something is wrong.

The monitor can be a) disabled and b) configured to allow for more slack, but that's all work-arounds: the robot should just execute the trajectory as ordered and the trajectory should not contain motions that the robot cannot perform. That is all part of the contract between the driver and the controller and should be respected by both sides.

Being explicit about contract breaches (ie: joint limit violations) is a good thing in such contexts.

gavanderhoorn, 2018-05-15:
> The behaviour I've observed in these sort of situations (our SIA20F on a rail) is that the safety systems of the controller will abort any motion as soon as it violates a limit, and the TP will show an error.

not remembering that ROS uses metres for everything and then asking your linear axis to move to +1000m in 1 second is quite a good way to observe this.

The rail moved only 1 cm before the controller figured out this was impossible and aborted the entire trajectory.

gavanderhoorn, 2018-05-15:
Note that other mfgs, such as KUKA, also don't do this (ie: scale). Typical errors seen when using RSI and commanding violating motions are `Command Gear Torque An` (with `n <- N`).

shaun-edwards, 2018-05-16:
I think it's fair for the robot to expect that the incoming trajectory is achievable  and if it's not, the user is responsible for taking corrective action.  The proposed change to alter the trajectory at the lowest level would be detrimental to any application where trajectory where timing was important.  Painting and welding come to mind.  Synchronizing with other equipment would also be dependent on trajectory timing (although I'm not sure I would recommend this practice).  

Traditional robot programs will sometimes alter their trajectories based on path constraints.  However, in these cases, the robot program is simply executing a pre-programmed path.  Once the path is "approved' by the operator, the execution will be consistent, even if not adhering exactly to the desired trajectory constraints. 

I would challenge your user to create achievable trajectories.  What within ROS or their software is specifying a bad trajectory?  It's much better to check the trajectory at a higher level, given that trajectory smoothing across multiple points would be needed.  Checking and correcting at the point to point level is not ideal.

gavanderhoorn, 2018-05-16:
@shaun-edwards: one use-case I could think of where the proposed automatic scaling does make sense is in material handling (ie: pick-and-place). You don't necessarily care how fast the robot is going, as long as it is as fast as it can go (as you want to minimise cycle times). There is no way to communicate that currently to MotoROS with `motoman_driver`.

However, setting too high velocities and then requiring MotoROS to scale back as needed does not seem like the correct way to achieve this (it's almost the opposite): it's like asking for something by *not* specifying it.

We could try to think of a nice way to achieve go-as-fast-as-possible without violating `JointTrajectory` or `JOINT_TRAJ_PT_FULL` and / or `ROS_MSG_MOTO_JOINT_TRAJ_PT_FULL_EX`.

gavanderhoorn, 2018-05-17:
https://github.com/ros-industrial/industrial_core/blob/d4a5df1339c6b0850453449443639bf4e227f984/industrial_robot_client/src/joint_trajectory_interface.cpp#L348-L356

gavanderhoorn, 2018-05-31:
@andreaskoepf: please post your last comment as a separate issue. This seems orthogonal to what is being discussed in this thread.

andreaskoepf, 2018-06-03:
(@gavanderhoorn I have to do a further analysis, I realized that on the MotoPlus side the Command-Position is already used, e.g. it probably is the problem I mentioned in https://github.com/ros-industrial/motoman/issues/217#issuecomment-393866928, e.g. MoveIt reads joint feedback as start-positon of next trajectory segment but the motor-controllers have not converged at that point -> command pos != feedback pos).

gavanderhoorn, 2018-06-03:
You could perhaps already open an issue. The fact that it was / is unclear to you what is going on means it's worth documenting.

It could be an 'investigative' issue.

gavanderhoorn, 2019-09-04:
I believe we reached a consensus about this @ted-miller (as in: don't auto-scale trajectories, but error out).

If you agree, could you please close this issue?

If there are any action items resulting from this discussion, could you open new issues for those?

closed 2019-09-05 by ted-miller