ros-industrial/motoman #218: Why is the root namespace specified for "/robot_enable" & "/robot_disable"?

andreaskoepf, 2018-06-01:
In `MotomanJointTrajectoryStreamer::init()` the handy `robot_enable` and `robot_disable` services are advertised, see [here](https://github.com/ros-industrial/motoman/blob/808a8c16d9120814c11438b8809cfb957bf3a70f/motoman_driver/src/joint_trajectory_streamer.cpp#L89-L91). Unfortunately the service name during the registration contains a '/' at the beginning and therefor overrides any namespace defined in a `<group>` element in a launch file, e.g. in Rosvita the launch-file generated for motoman driver looks like this:

```
<launch>
  <group ns="sda10d">
    <param name="robot_ip_address" type="str" value="10.0.0.1" />
    <node name="joint_state" pkg="motoman_driver" type="robot_state" />
    <node name="motion_streaming_interface" pkg="motoman_driver" type="motion_streaming_interface" />
    <node name="joint_trajectory_action" pkg="motoman_driver" type="motoman_driver_joint_trajectory_action" />
  </group>
</launch>
```

e.g. we use namespaces in order to allow multiple robots to be configured at once. The motoman actions and the `stop_motion` service are correctly placed under the specified namespace but robot_enable and robot_disable stay at the root namespace. Was the decision for the root-ns made deliberately (otherwise I could create a PR)?

gavanderhoorn, 2018-06-19:
This is indeed not as things should be, thanks for reporting this @andreaskoepf.

If you could provide a PR we'll review and merge.

This cannot be done in a bw-compatible way (users relying on those services being in the global namespace will not find them there any longer), but I'm willing to accept a breaking change as it's basically not correctly implemented right now.

gavanderhoorn, 2018-08-13:
Ping @andreaskoepf?