ros-industrial/motoman #223: Possible error with feedback velocity

ted-miller, 2018-06-21:
https://github.com/ros-industrial/motoman/blob/36ec447f3990e0f2313b95e24b40514443259cbd/motoman_driver/MotoPlus/CtrlGroup.c#L415-L428

Based on a report from another user using my test procedure, it appears that the correction for cross-axis coupling is being applied twice on DX200 and YRC1000. These newer controllers use the M-Register analog feedback function to read axis speeds. We think that the correction has already been applied to the values in the M-registers. So, the MotoROS code should not attempt to apply the correction.

@EricMarcil - Can you validate and fix this issue next week? If not, then I won't get around to it until July.

gavanderhoorn, 2018-06-21:
Should we assign this to @EricMarcil?

ted-miller, 2018-06-21:
Sure... but I can't get that to work.  Does he need to be added a maintainer of this repo?

I talked to Eric on the phone and he is going to look at this.

gavanderhoorn, 2018-06-21:
Hm, yes. He needs to be member of one of the groups that have access to this repository. Sorry about that.

(and we should actually do that)

gavanderhoorn, 2018-06-22:
off-topic, but I just invited @EricMarcil to the relevant teams for `motoman` and `motoman_experimental`.