ros-industrial/motoman #235: Failed to find topic_list parameter

Ranguet, 2018-08-01:
Hi i'am currently using a GP7 robot and YRC1000 controller, I followed all the steps from the tutorial, but I have a problem, when executing the roslaunch command i got this : 

`started roslaunch server http://ubuntu:44540/

SUMMARY
========

PARAMETERS
 * /controller_joint_names: ['joint_s', 'join...
 * /robot_ip_address: 192.168.255.1
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /
    joint_state (motoman_driver/robot_state)
    joint_trajectory_action (industrial_robot_client/joint_trajectory_action)
    motion_streaming_interface (motoman_driver/motion_streaming_interface)

auto-starting new master
process[master]: started with pid [9046]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 053d76dc-956d-11e8-b666-000c293ae231
process[rosout-1]: started with pid [9059]
started core service [/rosout]
process[joint_state-2]: started with pid [9066]
process[motion_streaming_interface-3]: started with pid [9077]
process[joint_trajectory_action-4]: started with pid [9086]
[ERROR] [1533115611.317375372]: Failed to find topic_list parameter
[ERROR] [1533115611.362236163]: Failed to find topic_list parameter
`
I check some helps, changing the joint names etc, but it seems that the urdf file is not changing and I am confused about it need to change or not.
Thank your for your help.

gavanderhoorn, 2018-08-01:
This could be an instance of #113.

You probably don't need to change anything, as long as the joints you list in `controller_joint_names` are present in your urdf.

Can you execute trajectories with your setup, or are you running into other problems?

Ranguet, 2018-08-01:
It's okay i just ignore the error message and it works now.
Thank you, the error message was a bit confusing sorry.

gavanderhoorn, 2018-08-01:
No need to apologise. As #113 states, the error is confusing, and you're right to open the issue.

Glad to hear things are working.

closed 2018-08-01 by Ranguet

Pranav186, 2020-05-28:
Hello, 
Hope this is the correct place to post this problem.
I am also getting the error - 
```
[ERROR] [1590678563.447733281]: Failed to find topic_list parameter
[ERROR] [1590678563.454831343]: Failed to find topic_list parameter
```
after running the command 
```
roslaunch motoman_driver robot_interface_streaming_fs100.launch robot_ip:=10.0.0.2
```

After waiting for a few minutes, there is a timeout error. This is the entire message-
```
started roslaunch server http://Robotics:45917/

SUMMARY
========

PARAMETERS
 * /robot_ip_address: 10.0.0.2
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    joint_state (motoman_driver/robot_state_bswap)
    joint_trajectory_action (industrial_robot_client/joint_trajectory_action)
    motion_streaming_interface (motoman_driver/motion_streaming_interface_bswap)

ROS_MASTER_URI=http://localhost:11311

process[joint_state-1]: started with pid [8970]
process[motion_streaming_interface-2]: started with pid [8971]
process[joint_trajectory_action-3]: started with pid [8977]
[ERROR] [1590678563.447733281]: Failed to find topic_list parameter
[ERROR] [1590678563.454831343]: Failed to find topic_list parameter
[ERROR] [1590678694.102028546]: Failed to connect to server, rc: -1. Error: 'Connection timed out' (errno: 110)
[ERROR] [1590678694.102028812]: Failed to connect to server, rc: -1. Error: 'Connection timed out' (errno: 110)

```

Also when i run ```rostopic echo joint_states```, I get the warning 
```WARNING: topic [/joint_states] does not appear to be published yet. ```

If i run the ```rostopic echo joint_states``` after the timeout,  the cursor keeps blinking and there is no output

I do not have experience with networking and ip addresses. I think there could be an issue with the connection.

I tried ```ping 10.0.0.2``` and there is a single line of output-
```PING 10.0.0.2 (10.0.0.2) 56(84) bytes of data.``` 
and the cursor keeps blinking

 The ip address 10.0.0.2 is the one found on the pendant

gavanderhoorn, 2020-05-28:
Please do not post your issue as a comment under an issue which has already been closed.

Open a new issue. If it turns out it's a duplicate of another issue, we'll close yours and redirect to the issue it is a duplicate of.

gavanderhoorn, 2020-05-28:
A quick comment though:

> `roslaunch motoman_driver robot_interface_streaming_fs100.launch robot_ip:=10.0.0.2`

You're not really supposed to start that `.launch` file by itself, but should use the `.launch` files provided by a robot support package.

See [wiki/Working with ROS-Industrial Robot Support Packages](http://wiki.ros.org/Industrial/Tutorials/WorkingWithRosIndustrialRobotSupportPackages) for an overview of how that would work.

lucky456pf, 2020-06-10:
> It's okay i just ignore the error message and it works now.
> Thank you, the error message was a bit confusing sorry.

hello,i'm also useing the ros  connect gp7,but my controller is yrc1000micro. And i'm useing ros kinect . i also have the proble "Failed to find topic_list parameter......" can you tell me how to fix it in detail？
 Thank your for your help.

gavanderhoorn, 2020-06-10:
Have you checked https://github.com/ros-industrial/motoman/issues/235#issuecomment-409515401?

If you have, and it still doesn't work, please post a new issue.