ros-industrial/motoman #237: Revert 564ad80 and force-push kinetic-devel?

gavanderhoorn, 2018-08-09:
Due to an unfortunate circumstance, 564ad80 was introduced a week ago on `kinetic-devel`.

This is a pure merge commit which merged `kinetic-devel` into itself. No new commits beside that one were introduced.

My suggestion would be to drop that commit from `kinetic-devel`.

Unfortunately that would mean we'd have to force-push to `kinetic-devel`, which may cause some issues for users that already have a clone (either a fresh one or recently updated) with that commit.

None of the public forks that appear on the [network](https://github.com/ros-industrial/motoman/network) appear to have caught up with their upstreams, so the inconvenience for those would appear to be minor.

I'm aware of the fact that force-pushing to a public repository like this is frowned upon, and I normally wouldn't suggest it, but I hope we caught it early enough so we can clean up the history while also minimising any inconvenience.

I'm mostly looking for opinions here from the maintainers: @shaun-edwards, @ted-miller.

ted-miller, 2018-08-09:
Since I caused the problem, I'm probably not the best person to give my suggestion... :)

But, I personally think that it should be left alone. If there is a potential risk of interfering with other users, then I think we should just ignore the small blip in the history. If anyone inspects the history, then they can see that it is an inconsequential commit. I would be more in favor of adding another commit, whose sole purpose would be to clarify the history and also remove the old binaries that I accidentally added.

In regard to the old binaries, I also think these are safe to ignore. First, they are clearly labeled with a version number in the file name. Plus, for the users who download the binaries directly, they will probably be using the Wiki, which links to the correct (v1.8.1) files.

shaun-edwards, 2018-08-09:
Is the concern having a clean history?  If so, I'm inclined to agree with @ted-miller.  The branch was broken for a bit,  it happens.   If there's is a larger concern,  please let me know.  I'll be the first to admit I don't fully understand what happened.

gavanderhoorn, 2018-08-09:
It's mostly about history.

Obsolete files were introduced and there is an unnecessary merge commit in there that is showing up in branch comparisons and on github.

Since it's only a single merge commit it should be relatively painless to remove, and will immediately result in a cleanup without any `git rm` or other actions needed.

gavanderhoorn, 2018-08-09:
The nr of users inconvenienced by this should be really minimal: you'd have to have created a clone between now and 8 days ago. Accordint to the repository statistics, there were about 10 unique cloners in that period (not counting clones by Travis runs).

gavanderhoorn, 2018-08-13:
With the merge of #236 this has now become slightly more complex.

So let's forget about this.

To prevent this in the future I've increased security settings on `kinetic-devel`.

closed 2018-08-13 by gavanderhoorn