ros-industrial/motoman #242: Failed to find topic_list parameter

hufman4, 2018-08-30:
[master.log](https://github.com/ros-industrial/motoman/files/2336640/master.log)
Hi Guys,
I have a problem when I try to connect to the DX200 driver.
When I start the file 'robot_interface_streaming_dx200.launch', I get the following error message:
```
ROS_MASTER_URI=http://localhost:11311

process[joint_state-1]: started with pid [13583]
process[motion_streaming_interface-2]: started with pid [13584]
process[joint_trajectory_action-3]: started with pid [13586]
[ERROR] [1535639556.481246527]: Failed to find topic_list parameter
[ERROR] [1535639556.490833696]: Failed to find topic_list parameter

```
Any idea what the problem might be?

Thanks in advance

gavanderhoorn, 2018-08-30:
Please see #235.

gavanderhoorn, 2018-08-30:
For future questions: please also always mention which ROS version you are using, how you installed the packages, controller version, which version of MotoROS you have installed and the specific robot variant that you are using.