ros-industrial/motoman #244: Precision of trajectory data fields

ted-miller, 2018-09-11:
We have found an issue with the precision of the data fields in a trajectory message. On the ROS/ROS-I side of things, all real values are treated as a 64-bit double. But, when the trajectory class gets serialized and sent to MotoROS, they are converted to a 32-bit float. 

This issue has caused at least one major problem. The 'time' field in the trajectory is used to calculate the size of each motion increment, effecting the robot's velocity. In this particular case, the user was intentionally running the robot very close to 100% of maximum velocity. All motion was done as a single trajectory for multiple hours. After the timestamp lost decimal precision over time, the MotoROS algorithm miscalculated the size of a motion increment. This caused the robot to exceed the speed limit, raise an alarm on the pendant, and halt the system while it was in production. This happened repeatably at regular time intervals.

For this specific case, one of my colleagues has determine that the 'time' field will begin to lose precision after 4.6 hours. 
[image]
For an immediate remedy, I am going to reject any trajectory command with a timestamp greater than 4.5 hours. But, this raises more questions about other possible problems caused by a loss of precision.

On the ROS-I side, it looks like it is controlled by the [FLOAT64 compile switch](https://github.com/ros-industrial/industrial_core/blob/kinetic-devel/simple_message/include/simple_message/shared_types.h#L65-L69). Why is this set to compile to 32-bit floats; is there a reason double isn't used? Did we (Yaskawa) decide to use float and force ROS-I to use it, or was it the other way around? Is this common for non-Motoman robots as well?

I'm wondering how we could make an update to support double's while still maintaining backward compatibility. Since there is no version-identifier in the message structure, I see two ways to do it.

1) Add a new command-type to switch to 64-bit mode. This will tell MotoROS to expect structures with 64-bit data fields (larger packet size). But it would require all future implementations to send this command message during startup.

2) Modify the existing message format to support a union of the two types of structures. I could add some detection in the structure so that I know what size packet to send/recv. But, I'm guessing this breaks the standard serializer class that ROS-I is using today.
[image]
 
Does anyone have some thoughts on the issue?

gavanderhoorn, 2018-09-11:
Just a quick comment -- but certainly not an 'excuse':

> All motion was done as a single trajectory for multiple hours.

This is far from a regular use-case, and I'm not surprised that this has not come up earlier. It's *very* uncommon for ROS applications to be able to precalculate trajectories of such length, mostly because they tend to include vision and other sensors in the motion planning process which are needed because of the dynamic environments that the application is targeted at.

It is of course not something we should not support, but I thought I'd add this here to provide some more context.

ted-miller, 2018-09-11:
I certainly agree that this is not normal. In fact, the countermeasure for this issue was to start a new trajectory whenever the robot came a brief stop.

But the fact that it happened once, means that it could happen again. And this particular issue was very difficult to diagnose, involved many people, and impacted customer production rates.

Plus, it makes me wonder if precision-loss could impact the 'position' field as well. Yaskawa does offer some high-precision robots.

So... if there's a way to fix the problem, then I think it should be addressed.

gavanderhoorn, 2018-09-11:
As to why float32 is the (apparent) default: Yaskawa controllers are exceptions, in that they mostly follow C typing for their internal variables.

A lot of industrial controllers serialise their floating point values to float32 (ie: 4 bytes) when transferring them over a wire and have no other representations available. This is why `float32` was chosen.

For those cases that float64 is available, the compile time constant that you found can be used to compile a version of the libraries that will assume and (de)serialise all floating point values to the 64bit variant.

We could consider compiling `motoman_driver` with this flag set. That would seem to take care of this issue.

Doing it wholesale at once is perhaps not a good idea though: as you write, there are no versioning fields available so detection of what a client (or a server) is using is impossible.

Another option could be to provide two sets of binaries for `motoman_driver`: one with the default float32 (de)serialisation, and another with float64 (de)serialisation.

> But the fact that it happened once, means that it could happen again. And this particular issue was very difficult to diagnose, involved many people, and impacted customer production rates.

I appreciate the inconvenience. I've run into similar issues with the native languages of certain robot manufacturers as well though. First thing to check in case of long trajectories (or great distances, etc) is to see whether the data representations can actually encode for those (but: captain hindsight, etc).

gavanderhoorn, 2018-09-11:
Perhaps @shaun-edwards can add some more context from the beginning of simple msg development.

ted-miller, 2018-09-11:
> Can you clarify what "high-precision robot" means? Are we talking millionth of degree precision? I agree that double precision could be better, but if I'm not mistaken, 6 significant digits available for angles seems to cover that.

I just checked the specs for our high precision robot. You're right, 6 significant digits is adequate for the position in radians. And while a 100m track is possible, I don't know that it is realistic. (But maybe possible on a robot controlled camera system...) So perhaps the 'position' field doesn't matter.

> A lot of other industrial controllers however serialise their floating point values to float32 (ie: 4 bytes) when transferring them over a wire and have no other representations available. This is why float32 was chosen: it was the most compatible representation for a large set of controllers that the protocol targeted.

So you're saying this is a ROS-I standard across the board and that adding 64-bit support would be a Motoman-specific branch?

shaun-edwards, 2018-09-14:
I don't know of anyone who has actually put any appreciable amount of runtime on the `float64` version.  It wasn't part of the original simple message library.  I believe if was added by someone for the UR...it's been a while.  

From the CMakeLists.txt, it does appear that we are testing that particular variation, so that's good.

If the `motoman_driver` uses `float` as it's internal data type, then increasing the precision of the messaging library will not have the desired effect.  I assume this is what @gavanderhoorn meant by testing the compatibility of the `motoman_driver`.

>1. Add a new command-type to switch to 64-bit mode. This will tell MotoROS to expect structures with 64-bit data fields (larger packet size). But it would require all future implementations to send this command message during startup.

>2. Modify the existing message format to support a union of the two types of structures. I could add some detection in the structure so that I know what size packet to send/recv. But, I'm guessing this breaks the standard serializer class that ROS-I is using today.

Of @ted-miller, two ideas above.  I believe the second one would not only cause problems for ROS-I, but also for anyone else who is using this interface.  The first idea appears to be backwards compatible.  On the ROS-I side, I imagine we could just use `sizeof` method to check the timestamp size and then use the appropriate command-type when sending the message.  It feels a bit hacky, but I think if backwards compatibility is a requirement, then we are stuck.

gavanderhoorn, 2018-09-14:
@shaun-edwards wrote:
>  I believe if was added by someone for the UR...it's been a while.

you mean kelsey for the c-api driver?

> If the `motoman_driver` uses `float` as it's internal data type, then increasing the precision of the messaging library will not have the desired effect.  I assume this is what @gavanderhoorn meant by testing the compatibility of the `motoman_driver`.

No, MotoROS uses encoder pulses internally, stored in `long`s.

What I meant with "compatibility of `motoman_driver`" is that we need to test whether all (de)serialisation codepaths in `motoman_driver` nodes properly use `FLOAT64` instead of the regular size.

Personally I would really like it if we could avoid making changes to messages or add hacks with `sizeof()`.

The cleanest way to do this would be to just add a new message type that uses `double` instead of `float` (so no `shared_real`s anywhere). We'd essentially use the msg ID as a version nr.

If we then make MotoROS accept *only* the new message ID and reject any other, outdated clients won't be able to command any motion and users would immediately become aware of the fact they need to fix something.

gavanderhoorn, 2018-09-14:
Note btw: migrating to `double` is not a fix. It's an example of enlarging the size of a datatype or moving to a wider datatype to make sure "our numbers fit in it". I know with `double` we get twice as many significant digits, but of course it'll only be a nr of time before someone tries to plan an entire week ahead in a single trajectory and we have the exact same problem again.

I know "640KB of memory should be enough for everyone", but it would probably be a good idea to add some checks to `motoman_driver` that capture these sort of extreme cases and at least warn, or better yet perhaps even just reject them.

shaun-edwards, 2018-09-14:
>The cleanest way to do this would be to just add a new message type that uses double instead of float (so no shared_reals anywhere). We'd essentially use the msg ID as a version nr.

You're right.  This is better.

ted-miller, 2018-09-14:
Migrating to double is a fix for possible "unknown consequences" due to loss of precision.

In regard to the trajectory time, I am going to modify MotoROS to reject trajectories with a timestamp greater than xxxx. I will return a new error code. This will fix the immediate problem.

> If we then make MotoROS accept only the new message ID and reject any other, outdated clients won't be able to command any motion and users would immediately become aware of the fact they need to fix something.

Personally, I disagree with this. I'm a fan of maintaining as much backward compatibility as possible. I'd suggest that we push the new 64 bit messages as the new standard, but I don't want to break any existing software installations.

gavanderhoorn, 2018-09-14:
I'm not sure where the issue with bw-compatibility is.

We'd update MotoROS and update `motoman_driver`.

I don't see a way to go to FLOAT64 without doing both of those things.

gavanderhoorn, 2018-09-14:
@ted-miller wrote:
> Migrating to double is a fix for possible "unknown consequences" due to loss of precision.

And for the problem with the `time` field. Its units are seconds, and with 15 significant digits it would encode for a lot of seconds.

Another option could be to do away with floating point representations for time completely, and use a 64 bit integer that counts milliseconds. That would give us a maximum trajectory length of 18,446,744,073,709,551 seconds without any loss of precision (and if we decide that it should be unsigned).

ted-miller, 2018-09-14:
> I'm not sure where the issue with bw-compatibility is.

Up to now, we haven't had any need to link the versions. So, when people call me with support questions, my first recommendation is to update MotoROS. This ensures that their issue hasn't already been resolved on my end. 

> I don't see a way to go to FLOAT64 without doing both of those things.

If someone (who has already developed their PC software) thinks that something is buggy in the robot, then the first thing they do is update MotoROS.  IMO, folks generally don't want to update components that are more tightly integrated into their proprietary PC software.

gavanderhoorn, 2019-09-04:
An artificial limit on maximum trajectory length was introduced in #256.

Closing this.

closed 2019-09-04 by gavanderhoorn