ros-industrial/motoman #249: Motoman Kinetic - Issues Building from source

hdemel-PN, 2018-10-03:
Hi,

We are trying to install the kinetic driver package. 

As I see it, we are to compile and build 3 packages for this to be successful. These are the packages located in:

- motoman
- motoman_driver
- motoman_msgs

I can install the "motoman" and "motoman_msgs" packages without issue, when trying to run cmake on motoman_driver I receive the attached error.

[image] 

ros-kinetic-genmsg has been installed, what could be causing this?

gavanderhoorn, 2018-10-03:
Can you please show us how you builld everything exactly? Please copy-paste *all* commands you use.

Note: these are Catkin packages, meant to be built in a ROS workspace. See [Creating a workspace for catkin](http://wiki.ros.org/catkin/Tutorials/create_a_workspace) for some high-level information about creating such a workspace.

In general the procedure would be:

 1. create a Catkin workspace
 1. `git clone ..` the repository into the `src` space
 1. install all dependencies using `rosdep` (see [252478](https://answers.ros.org/question/252478) on ROS Answers for the general procedure)
 1. build the workspace
 1. `source /path/to/workspace/devel/setup.bash`

From your screenshot I get the impression that you're trying to follow the regular `mkdir build; cd build; cmake ..; make` workflow.

That *can* work, but not without some additional steps.

gavanderhoorn, 2018-10-03:
As I don't believe this is an issue with the pkgs in this repository I'm going to close the issue.

Feel free to keep commenting on the issue of course.

closed 2018-10-03 by gavanderhoorn

hdemel-PN, 2018-10-03:
Hey thanks for your quick reply. You are totally correct, I used cmake to build from source. I wasn't aware that a different approach was required at the time.

Your instructions seem to have provided me with some success. If you don't mind reviewing:

- create a Catkin workspace
         - $ source /opt/ros/melodic/setup.bash
         - $ mkdir -p ~/catkin_ws/src
         - $ cd ~/catkin_ws/
         - $ catkin_make
         - $ source devel/setup.bash
         - $ echo $ROS_PACKAGE_PATH /home/probotix/catkin_ws/src:/opt/ros/kinetic/share

- git clone .. the repository into the src space
        - cd ~/catkin_ws/src
        - git clone www.....

- install all dependencies using rosdep (see 252478 on ROS Answers for the general procedure)
      - cd .. 
      - rosdep install --from-paths src --ignore-src --rosdistro kinetic -y -r --os=ubuntu:xenial
      - cd build
      - make
      - make install

- build the workspace
    - $ echo $ROS_PACKAGE_PATH /home/probotix/catkin_ws/src:/opt/ros/kinetic/share

- source /path/to/workspace/devel/setup.bash
   - $ source devel/setup.bash

These are the exact commands used, do these look correct? No errors were generated during the process and I am able to successfully launch the yrc1000 launch file. 

Does this look accurate to you?

gavanderhoorn, 2018-10-03:
The way you build your workspace is still not correct.

You should use `catkin_make` or `catkin build`.

If you're getting into ROS, then try to avoid using the `mkdir build; cd build; cmake ..; make` workflow.

gavanderhoorn, 2018-10-03:
The fact that `roslaunch` cannot find the nodes is most likely because you source the wrong `setup.bash` file: you're sourcing the one of the workspace, while you should source the one in your `build` directory -- but that is only because you used the "wrong" build steps.

hdemel-PN, 2018-10-03:
@gavanderhoorn your help is greatly appreciated.

I have rebuilt and reinstalled using catkin_make and catkin_make install in the root of catkin_ws. Now once started, I get an error saying "failed to find topic_list parameter". 

rosservice call robot_enable still fails as well.

> The fact that roslaunch cannot find the nodes is most likely because you source the wrong setup.bash file: you're sourcing the one of the workspace, while you should source the one in your build directory -- but that is only because you used the "wrong" build steps.<

I don't have a setup.bash in catkin_ws/build so I have tried "source ~/catkin_ws/install/setup.bash". With this, I still get the topic_list error and I am still unable to call the robot_enable service.

Again, thanks for your patience @gavanderhoorn

gavanderhoorn, 2018-10-03:
If you haven't yet, I would suggest you go through the general ROS tutorials. They will teach you how to interact with a workspace and how to create and build packages. The questions you ask tell me that you are not familiar enough with the workflow to be able to figure out what is caused by missing things in the workspace or what is caused by other things.

I write this not because I don't want to help you, but because it will help *you* tremendously to be able to figure out some things for yourself, instead of having to rely on others.

As to your problems:

> Now once started, I get an error saying "failed to find topic_list parameter".

That is because you cannot just start the driver node in isolation, it requires certain configuration parameters to be present.

I'm going to need more information about the robot that you have connected to your YRC1000 and how many motion groups you have. If you only have a single motion group, you can ignore the error about the `topic_list` parameter (see #113 and #235 for related issues).

> I don't have a setup.bash in catkin_ws/build so I have tried "source ~/catkin_ws/install/setup.bash"

I wasn't referring to `setup.bash` in `catkin_ws/build`, but it doesn't matter.

Also: when developing, it's customary to *not* use the `install` space, but to `source` the file in `devel`.

hdemel-PN, 2018-10-03:
Hi @gavanderhoorn I will definitely try giving the tutorials another go. We're working with our local Yaskawa office on developing this environment and road-mapping some applications for it so our timeline is a little compressed.

Have I sourced the correct file for the installation of the motoman driver? Is there any documentation about the configuration parameters required to clear the errors related to "topic_list" that you could direct me towards?

Again, thanks for your help. Your patience is very much appreciated.

gavanderhoorn, 2018-10-03:
Please provide me with a description of your system.

Without that I cannot help you.

hdemel-PN, 2018-10-03:
Hey @gavanderhoorn, we have a YRC1000micro and a Moto-mini arm. I am not sure what you mean by motion groups. 

Is there any other information I can provide you?

Also, I have managed to invoke robot_enable. I am having issues with rosrun not recognizing motoman_driver with the command: rosrun motoman_driver move_to_joint.py "[J1, 4, -0.81, -0.826, -0.018, 1.65, 0.0]"

[rospack] Error: package 'motoman_driver' not found

hdemel-PN, 2018-10-04:
@gavanderhoorn Thanks a lot for your help. 

@EricMarcil is working on the URDF for us.

To avoid the issue of having to source the setup.bash file everytime, I included the path in my ~/.bashrc file. 

I think everything is good to go otherwise, once the URDF is ready we will be ready to test with the actual robot. The robot_enable service works now and will enable the servos on the controller.

Thanks again for your help.

tedboss, 2019-06-05:
Hi,
We are working on a project that requires ROS on YRC1000micro controller, but as I found on ROS wiki and Motoman drivers, there is no YRC1000micro support

Fortunately, I found this thread, I would like to ask if you have launch successful ROS on YRC1000micro (motomini)??

Thank you very much

gavanderhoorn, 2019-06-06:
@tedboss wrote:
> there is no YRC1000micro support

that's not actually correct: the micro is fully supported. It's only the `wiki/motoman_driver` page that doesn't list it explicitly.

The [installation tutorial](http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer) lists it and links you to the MotoROS version to use.

The wiki page should be updated. I've logged #278 to track that.

marcochen11, 2020-01-15:
@hdemel-PN 
Hi! I have the same problem as you did.
I'm using yrc1000u controller and GP7 robot. I cannot invoke robot_enable command either. How did you solve is at that time?

Code from terminal listed below:
`
marco@marco-Blade:~/catkin_ws$ rosservice call robot_enable
ERROR: Service [/robot_enable] is not available.
`

gavanderhoorn, 2020-01-15:
@marcochen11: please don't post a comment about an unrelated issue on an already closed thread.

Please post this as a new issue on the tracker and we'll take a look.