ros-industrial/motoman #257: Message Reply conflict

zhubeilife, 2018-12-10:
MotoROS will send back MOTO_MOTION_REPLY each time it receives a joint trajectory message or a motoman motion control command. If I send joint trajectory message and motoman motion control command one by one. And I receive two MOTO_MOTION_REPLY message, but it seems the message is received by wrong message handler/receiver.

gavanderhoorn, 2018-12-10:
I'm having a bit of difficulty following your description.

Could you perhaps explain the flow of traffic a bit more?

Are there multiple connections to the controller?

zhubeilife, 2018-12-13:
Sorry for my bad descriptions.

For example we send message **A**(joint trajectory message ) and the control will send back **B** (MOTO_MOTION_REPLY) to tell the controller receice the trajectory. 
We send message C (Motion control command), the robto will send back D (MOTO_MOTION_REPLY) to tell how the command works or not.

Now we send A and B one by one. the Controller will reply C and D.
But it seems some C is received by B, and D is receive by A.

gavanderhoorn, 2019-01-11:
> Thanks for the help, we found the problem. We are directly using some motion_ctrl functions to communicate with the controller. But when we are sending some trajectory the JointTrajectoryStreamer streaming Thread will also wait for the MOTO_MOTION_REPLY message when it sends a trajectory. Without some lock, the two message with mess up because they are using the same TCP connections.

Yes, that would be a problem, yes.

Communication over the same socket is serialised (on the ROS side) by locks (such as [this one](https://github.com/ros-industrial/motoman/blob/80c529979de52bb6eed5988d1e3076ae9abc131e/motoman_driver/src/joint_trajectory_streamer.cpp#L404)). If you make use of the same resource, you'll have to take this locking into account.

closed 2019-01-11 by zhubeilife