ros-industrial/motoman #258: Motoman Driver Error: Failed to init message, buffer size too small: 3

mbraun89, 2019-01-31:
Hi,

for safety reasons I want to call /robot_disable service, directly after an emergency stop occurred, to make shure, that the robot doesn't start to move after the emergency stop is reseted. This works properly 95% of the times, but sometimes I can not call /robot_disable service and get the following error message:

ERROR [/tmp/binarydeb/ros-kinetic-simple-message-0.6.0/src/simple_message.cpp:107(simple_message::SimpleMessage::init) [topics: /rosout] Failed to init message, buffer size too small: 3

Unfortunately I am not able to reproduce, under what circumstances this error occurres exactly, but it seems to be only while executing trajectories and not when the robot stands still. Is there an option to increase the buffer size, or can I be sure that the robot doesn't accept motion commands, after an emergency stop reset, regardless of whether calling /robot_disable service or not.

gavanderhoorn, 2019-02-01:
Do I understand you correctly that you are calling `robot_disable` while a trajectory is being executed (but execution (on the controller) is stopped/paused by the emergency stop)?

mbraun89, 2019-02-01:
Yes, that is right. I want to ensure that no motion is possible after somebody resets the emergency stop. The szenario is, that emergency stop button is pressed and released directly. I saw in the code, that clearing the queues etc. is done, when robot_disable is called. That is why I call robot disable directly after RobotStatus/e_stopped becomes 1, to get deterministic behaviour, when the emergency stop is resetted.
I made a few tests yesterday and it turns out, that everything works fine, when I wait for 1 second, after the e_stop occurres before calling robot_disable.

EricMarcil, 2019-02-01:
Hi,
I'm not sure about the cause of the error but as far as safety goes, it is safe.
When the e-stop is pressed the playback stops and the motion buffer is flushed out.
The motion can only occur while being in playback and the robot job (INIT_ROS) is on a WAIT instruction.

EricMarcil, 2019-02-01:
Thinking more about it, I'm wondering if the issue is that internally when there is an alarm the motion thread locks the motion queue and clears it.  This may prevents the motion server thread from processing properly the request "robot_disable" which also tries to do the same thing and times out.  By adding the 1 sec delay, you are probably allowing time for the internal process to release the lock on the motion queue.

mbraun89, 2019-02-04:
Hi,
I tried to reproduce the error. The error occurs when I call /robot_disable while the robot is in motion, regardless whether the e-stop is pressed before or not. Again, this happens just about 5% of the times. 
I get the following error messages:
[ERROR] [1549268967.943485504]: Failed to Stop Motion: Busy (1) : Unknown (0)
[ERROR] [1549268967.966164947]: Failed to initialize message
[ERROR] [1549268967.966230433]: Failed to send MotionCtrl message
[ERROR] [1549268967.966255459]: Failed to send TRAJ_MODE command
[ERROR] [1549268967.966278293]: Motoman robot was NOT disabled. Please re-examine and retry.

The robot stands still, so this works. But I can not call /robot_enable service any more.  I need to restart the motion_streaming_interface to re-enable the robot movement.
Unfortunately sometimes the /motion_streaming_interface just stops working without error messages. The /joint_trajectory_action node continues and detects the timeout, because the trajectory can not be completed:
[ERROR] [1549269746.100690496]: Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 0.798293 seconds). Stopping trajectory.

So I think, I need to stop all movement, before I can call /robot_disable, right?

yszheda, 2020-02-26:
Hi,

Recently I also the encountered the similar problem:

```
Failed to initialize message
Failed to send START_TRAJ_MODE command
Motoman robot was NOT enabled. Please re-examine and retry
```

I find a problem in the motoman driver: both `MotomanJointTrajectoryStreamer` and `MotomanMotionCtrl` use the same `industrial::smpl_msg_connection::SmplMsgConnection` object, and the `SmplMsgConnection` class is not thread safe. When the `MotomanMotionCtrl` calls `SmplMsgConnection::sendAndReceiveMsg` to send the START_TRAJ_MODE command, the `streamingThread` in the `MotomanJointTrajectoryStreamer` might also call `SmplMsgConnection::sendAndReceiveMsg` to send the JointTrajPtFullMessage. So when the two threads both call `SmplMsgConnection::receiveMsg` to read the same message from the same socket fd, one of them might get less bytes than it requires. I think we could either protects the `SmplMsgConnection` object and make `SmplMsgConnection::sendAndReceiveMsg` atomic, or just use different socket fds for `MotomanJointTrajectoryStreamer` and `MotomanMotionCtrl` to avoid contention.

gavanderhoorn, 2020-02-27:
There does appear to be a possibility for concurrent usage of the socket.

I believe this was introduced in #155, which added the enable/disable service.

Before that PR, even though `SmplMsgConnection::sendAndReceiveMsg` is not thread-safe, the control flow through `MotomanJointTrajectoryStreamer` never led to any issues, as it was "impossible" for two concurrent usages of the traj streamer socket to occur.

With the introduction of the enable/disable service, this changed, as those services may be called at any time, even when the socket is being used by the traj streamer, as you write.

`simple_message` is not thread-safe, but this is in purpose (or: it's left up to the user of that library to make sure no concurrent access of shared resources could occur). So I'd rather not add anything there.

What we could do is use the `boost::mutex` that is used in the "streaming thread" to serialise access to the socket object ([this one](https://github.com/ros-industrial/motoman/blob/43be182c9cb4f62806e479b985965886c8603333/motoman_driver/src/joint_trajectory_streamer.cpp#L404)).

That mutex is only grabbed whenever the streaming thread wants/needs to send and/or receive something over the socket, and grabbing it from the enable/disable service callbacks should prevent concurrent use of the socket.

Would you be able to try this and report whether it works?

yszheda, 2020-02-29:
Hi gavanderhoorn,

Thanks for the suggestion!

By looking at the following captured packets, now I can confirm that I encounter the concurrency problem:

[image]

And I write the following program which can be served as the minimal testcase to reproduce my problem:

```cpp
#include "rclcpp/rclcpp.hpp"
#include <iostream>
#include <string>
#include <chrono>
#include <thread>
#include <mutex>
#include <condition_variable>
#include "simple_message/socket/tcp_server.h"
#include "simple_message/socket/tcp_client.h"
#include "simple_message/simple_message.h"
#include "simple_message/log_wrapper.h"
#include "simple_message/messages/joint_traj_pt_full_message.h"
#include "motoman_motion_ctrl_message.h"
#include "motoman_motion_reply_message.h"

using industrial::simple_message::SimpleMessage;
using motoman::simple_message::motion_reply::MotionReply;
using motoman::simple_message::motion_reply::MotionReplyResult;
using motoman::simple_message::motion_reply_message::MotionReplyMessage;
using motoman::simple_message::motion_ctrl::MotionControlCmds::MotionControlCmd;
using industrial::simple_message::ReplyTypes::ReplyType;
using industrial::simple_message::StandardMsgTypes::StandardMsgType;

namespace CommTypes = industrial::simple_message::CommTypes;
namespace ReplyTypes = industrial::simple_message::ReplyTypes;

int main(int argc, char **argv)
{
  rclcpp::init(argc, argv);

  bool add_lock = false;
  if (argc > 1) {
    add_lock = (atoi(argv[1]) > 0);
  }
  std::cout << "Add lock to protect TcpClient: " << add_lock << std::endl;

  std::mutex mutex;
  std::condition_variable cond_var;

  std::string ip = "127.0.0.1";
  int port = 50111;

  auto server_thread = std::thread([port, &cond_var]() {
    industrial::tcp_server::TcpServer server;
    server.init(port);

    cond_var.notify_all();

    rclcpp::Rate loop_rate(10.0);
    while (rclcpp::ok())
    {
      if(!server.isConnected())
      {
        std::cout << "Server waiting for connection." << std::endl;
        if(server.makeConnect())
        {
          std::cout << "Server made connection." << std::endl;
        }
        else
        {
          std::cout << "Server failed to make connection." << std::endl;
          continue;
        }
      }

      SimpleMessage msg, reply;
      server.receiveMsg(msg);

      MotionReply mr_msg_data;
      mr_msg_data.init(0,
                       0,
                       StandardMsgType::JOINT_TRAJ_PT_FULL,
                       // MotionControlCmd::START_TRAJ_MODE,
                       MotionReplyResult::SUCCESS,
                       0,
                       0.0);
      MotionReplyMessage mr_msg;
      mr_msg.init(mr_msg_data);
      mr_msg.toReply(reply, ReplyType::SUCCESS);
      server.sendMsg(reply);

      loop_rate.sleep();
    }
  });

  {
    std::unique_lock<std::mutex> lock(mutex);
    cond_var.wait(lock);
  }

  industrial::tcp_client::TcpClient client;
  if (!client.init(strdup(ip.c_str()), port))
  {
    std::cerr << "Can't connect to " << ip << ":" << port << std::endl;
    return 1;
  }
  client.makeConnect();
  std::cout << "Connect to " << ip << ":" << port << std::endl;

  auto request_thread_1 = std::thread([&client, &add_lock, &mutex]() {
    industrial::joint_traj_pt_full_message::JointTrajPtFullMessage jtpf_msg;
    industrial::simple_message::SimpleMessage current_msg;
    jtpf_msg.toRequest(current_msg);

    industrial::simple_message::SimpleMessage request, response;
    request.init(current_msg.getMessageType(),
        CommTypes::SERVICE_REQUEST,
        ReplyTypes::INVALID,
        current_msg.getData());

    motoman::simple_message::motion_reply_message::MotionReplyMessage ctrl_reply;

    std::cout << "[thread 1] sendAndReceiveMsg start." << std::endl;

    if (add_lock)
    {
      mutex.lock();
    }
    auto is_msgs_sent = client.sendAndReceiveMsg(request, response, true);
    if (add_lock)
    {
      mutex.unlock();
    }

    if (!is_msgs_sent)
    {
      std::cerr << "[thread 1] sendAndReceiveMsg failed." << std::endl;
      return;
    }

    if (!ctrl_reply.init(response))
    {
      std::cerr << "[thread 1] Failed to initialize MotionReplyMessage." << std::endl;
      return;
    }

    std::cout << "[thread 1] success." << std::endl;
  });

  auto request_thread_2 = std::thread([&client, &add_lock, &mutex]() {
    motoman::simple_message::motion_ctrl_message::MotionCtrlMessage ctrl_msg;
    motoman::simple_message::motion_reply_message::MotionReplyMessage ctrl_reply;

    industrial::simple_message::SimpleMessage request, response;
    ctrl_msg.toRequest(request);

    std::cout << "[thread 2] sendAndReceiveMsg start." << std::endl;

    if (add_lock)
    {
      mutex.lock();
    }
    auto is_msgs_sent = client.sendAndReceiveMsg(request, response, true);
    if (add_lock)
    {
      mutex.unlock();
    }

    if (!is_msgs_sent)
    {
      std::cerr << "[thread 2] sendAndReceiveMsg failed." << std::endl;
      return;
    }

    if (!ctrl_reply.init(response))
    {
      std::cerr << "[thread 2] Failed to initialize MotionReplyMessage." << std::endl;
      return;
    }

    std::cout << "[thread 2] success." << std::endl;
  });

  request_thread_1.join();
  request_thread_2.join();

  rclcpp::shutdown();
  server_thread.join();

  return 0;
}
```

Protecting the critical section `SmplMsgConnection` is also my solution and it works on our ros2 version of  `motoman_driver`. I also write the patch for kinetic and will create a PR for it.

commit 44599376 (2021-12-15): driver: protect SmplMsgConnection with a mutex (fix #258)
closed 2021-12-15 by gavanderhoorn