ros-industrial/motoman #265: Where to find motoman_mh5_moveit_config ?

kctoayo88, 2019-02-15:
Hello, I am a beginner of ROS.

My robot is motoman mh5, and I found some helpful resources in this repository.
Then I want to use moveit! software to control my robot arm, but I can't find the moveit_config files of mh5 inside.
I searched this topic (#176 )  and don't know it's exist or not.

Hope you could help me to solve this problem,
Thanks a lot.

EricMarcil, 2019-02-15:
Hi @kctoayo88 

We've stopped including package for MoveIt a while back because it was just growing the repo and most people will need to create their own package to include other elements of the robot environment.
They are easy to make and there are plenty of tutorial online.
You can refer to the Ros Industrial Training to get you started:
[https://industrial-training-master.readthedocs.io/en/latest/_source/session3/Build-a-Moveit!-Package.html]

gavanderhoorn, 2019-02-15:
As @EricMarcil already explained why there is no MH5 MoveIt config pkg available (or at least: not from these repositories).

I'll close this as I believe we've sufficiently addressed the question.

Feel free to keep commenting on the issue of course.

closed 2019-02-15 by gavanderhoorn