ros-industrial/motoman #267: motoman sda10f [ERROR] [1531987897.956109]: Start and End position joint_names mismatch

khalilibrahim, 2019-02-21:
When, I sent position for each joint of motoman sda10f like
`rosrun motoman_driver move_to_joint.py "[1.2114869356155396, 1.9079313278198242, 1.724528431892395, 0.7935540676116943, 1.1871036291122437, 0.25184622406959534, 0.16729415953159332]" `
results in:

```
[ERROR] [1531987897.956109]: Start and End position joint_names mismatch
[ERROR] [1531987897.957380]: Unable to move to commanded position. Aborting.
Traceback (most recent call last):
  File "~/catkin_ws/src/motoman-kinetic-devel/motoman_driver/src/move_to_joint.py", line 136, in <module>
    main(sys.argv[1:])
  File "~/catkin_ws/src/motoman-kinetic-devel/motoman_driver/src/move_to_joint.py", line 130, in main
    move_to_joint(end_pos, duration)
  File "~/catkin_ws/src/motoman-kinetic-devel/motoman_driver/src/move_to_joint.py", line 70, in move_to_joint
    traj = build_traj(get_cur_pos(), end_pos, duration)
  File "~/catkin_ws/src/motoman-kinetic-devel/motoman_driver/src/move_to_joint.py", line 25, in build_traj
    raise
TypeError: exceptions must be old-style classes or derived from BaseException, not NoneType
```
anyone help me

gavanderhoorn, 2019-02-28:
I've logged an issue for the `TypeError` (#270), but I'm not sure this would ever work: an SDA robot has at least 2*7 joints and you only supply 7 values.

closed 2019-04-11 by khalilibrahim

gavanderhoorn, 2019-04-11:
Was this resolved, or are you no longer interested in it?

khalilibrahim, 2019-04-11:
Dear  G.A. vd. Hoorn,

Yes I solved now

Thank you.
Best regards,

Khalil Ibrahim, PhD
Mechanical Engineering Department
Faculty of Engineering, Assiut University
=========================
Address: Mechanical engineering department,Faculty engineering, Assiut University.
E-mail: khalil.ibrahim77@gmail.com,  khalil.ibrahim@aun.edu.eg.
Mobile:  +201155977599, +201147014783.
Tel.Internal: 1785.
Fax: +2-2335572.

gavanderhoorn, 2019-04-11:
If you did, it would be good to know how.

What changes did you make?

khalilibrahim, 2019-04-11:
I need to use STOMP planner insted of the OMPL in my package motoman_sda10f
I follow all instruction in https://github.com/ros-industrial/industrial_moveit but i fail to loading planner STOMP with my package.

Thank you.
Best regards,

Khalil Ibrahim, PhD
Mechanical Engineering Department
Faculty of Engineering, Assiut University
=========================
Address: Mechanical engineering department,Faculty engineering, Assiut University.
E-mail: khalil.ibrahim77@gmail.com,  khalil.ibrahim@aun.edu.eg.
Mobile:  +201155977599, +201147014783.
Tel.Internal: 1785.
Fax: +2-2335572.

gavanderhoorn, 2019-04-11:
How is this related to the issue you reported here about the `move_to_joint.py` script?