ros-industrial/motoman #269: motoman_mh6_support package?

npd26, 2019-02-28:
is anyone have idea about motoman_mh6_support package for ros-i?

gavanderhoorn, 2019-02-28:
There is no support for the MH6 variant in this repository or `motoman_experimental` right now.

I'm also not aware of any community contributed packages, but I also haven't searched for them.

gavanderhoorn, 2019-02-28:
I've labelled this an *enhancement*, but have not put it on the todo list for any milestone.

If you end up creating a support pkg for that variant yourself, a PR contributing it would be very much appreciated.

npd26, 2019-02-28:
I just found this link, 
https://github.com/sinaiaranda-CIDESI/motoman_mh6-10/tree/master/motoman_mh6_support

can I use it directly or any changes needed? specially for ROS-I?

gavanderhoorn, 2019-02-28:
Looks like it's just a bare urdf exported from SolidWorks using the SW2URDF plugin, but it can probable be used. I assume @sinaiaranda-CIDESI is using it, so perhaps he can comment on the quality/completeness of it.

Some things to check and/or change:

 - joint limits
 - incorrect inertias & masses (most likely)
 - not a xacro (so can't be composed)
 - no collision meshes
 - doesn't follow conventions (naming, file & directory layout, frame & link orientations)
 - configuration of motion group & joint data

npd26, 2019-02-28:
how can I use this to make compatible with ROS-I?

https://github.com/ros-industrial/motoman_experimental/tree/kinetic-devel/motoman_robot_pkg_gen

gavanderhoorn, 2019-02-28:
You can't. That's not what that tool does.

You'll have to manually update the MH6 package.

sinaiaranda-CIDESI, 2019-02-28:
Hi,
I use this package without problem. Just have the joint limit about the calibration pulse of the robot. You can open the demo moveit and see the home position and calibrate the robot pulses  in this position or update the package.

EricMarcil, 2019-02-28:
I saw in the URDF that some of the values are slightly off, probably not significant but if you make a package for distribution you might want to use the following values which are for a MH6-A00.  Note there are variation on that robot model.  So you might want to double check the model you have by looking at the identification plate on the robot.

DX100 parameters for robot model MH6-A00

Axis, Theta[rad], D[mm], A[mm], Alpha[rad], Variable, Ratio[pulse/rad or pulse/mm], Motion Limit Min [rad or mm], Motion Limit Max [rad or mm], Speed Limit [rad/s or mm/s], Max Acc/Dec Torque[Nm], Rated Motor Torque[Nm]
1(S), 0.000000, 0.000, 150.000, 1.570796, theta, 82239.5216, -2.967065, 2.967065, 3.839723, 329.28, 2.86
2(L), 1.570796, 0.000, 614.000, 0.000000, theta, -74502.7025, -1.570802, 2.705258, 3.490652, 668.85, 5.39
3(U), 0.000000, 0.000, 155.000, 1.570796, theta, 78879.7363, -1.448636, 3.141593, 3.839731, 191.59, 2.86
4(R), 0.000000, 640.000, 0.000, -1.570796, theta, -32594.9323, -3.141593, 3.141593, 7.155836, 20.11, 0.32
5(B), -1.570796, 0.000, 0.000, 1.570796, theta, 47206.4537, -0.785401, 3.926984, 7.155844, 26.15, 0.32
6(T), 0.000000, 95.000, 0.000, 0.000000, theta, -24382.7026, -6.283184, 6.283184, 10.646523, 13.72, 0.32

Pulse Position, 0,0,0,0,0,0
Robot Position, 790.000, 0.000, 674.000, 180.000000, 0.000000, 0.000000

Pulse Position, 10000,10000,10000,10000,10000,10000
Robot Position, 889.428, 80.429, 670.020, 159.234894, -3.251444, 33.010005

@npd26 you can email me at eric.marcil@motoman.com and I can send you the CSV file with the above values or for a different model.

closed 2019-05-08 by npd26