ros-industrial/motoman #272: Joint velocity feedback value's sign don't change

zhubeilife, 2019-03-26:
Hi, 

    We are using MotoROS1.7.0 with YRC1000 controller and GP25. It seems the Joint feedback value's sign doesn't change. Like S joint velocity is always position, and L joint always is negative. 
    We test it with moving the joint with a different direction and get velocity by Motoman driver or Wireshark.
    Is it related to the motors code part?

Thanks a lot.

gavanderhoorn, 2019-03-26:
@zhubeilife: could you please provide a wireshark capture with some traffic showing this?

gavanderhoorn, 2019-03-26:
@ted-miller: could this be a sign extension issue somewhere around [here](https://github.com/ros-industrial/motoman/blob/7de25115d7a081bf285db98972ab311c23660b65/motoman_driver/MotoPlus/CtrlGroup.c#L408-L413)?

gavanderhoorn, 2019-03-26:
@zhubeilife:

> We are using MotoROS1.7.0 with YRC1000 controller and GP25

any reason you're not using one of the `1.8.x` versions ([here](https://github.com/ros-industrial/motoman/tree/80c529979de52bb6eed5988d1e3076ae9abc131e/motoman_driver/MotoPlus/output/YRC1000/MotoRosYRC1_))?

The velocity code has seen quite some changes in `1.8.x`. Please test with `1.8.1` first and let us know whether you still encounter this problem.

zhubeilife, 2019-03-26:
> @zhubeilife:
>
>
> any reason you're not using one of the `1.8.x` versions [...]

Because we modify some part of the MotoROS code to fit our needs.

OK, I will first test with the new version.

gavanderhoorn, 2019-03-26:
> Because we modify some part of the MotoROS code to fit our needs.

For future support requests: *always* mention this sort of information.

We cannot help you if you don't tell us this.

zhubeilife, 2019-03-26:
> > Because we modify some part of the MotoROS code to fit our needs.
> 
> For future support requests: _always_ mention this sort of information.
> 
> We cannot help you if you don't tell us this.

Sorry for the  inconvenience

gavanderhoorn, 2019-03-28:
@zhubeilife: did `1.8.1` also exhibit the problem?

zhubeilife, 2019-03-29:
Yes, we test with V1.8.1 with below result. We move a single joint speratly and it seems the feedback speed always be positive. 

[image]

And I look into the code, the value read from the register is an unsigned type.

zhubeilife, 2019-03-29:
Today, I also test it with GP8 YRC1000 with version 1.7.0. I add directly print the register value in MotoROS code and connect with the controller using telnet. It seems the register value didn't change with different directions.

gavanderhoorn, 2019-03-29:
That would seem to suggest a sign issue, indeed.

@ted-miller: any idea?

ted-miller, 2019-03-29:
MotoROS gets the speed values directly from M Registers which are populated by the controller's system software. As zhubeilife pointed out, the register values will always be positive. So, there is no way for MotoROS to determine the direction of the motion.

I think (I'm not certain) that the method used by DX100 and FS100 does actually indicate direction. [See here](https://github.com/ros-industrial/motoman/blob/kinetic-devel/motoman_driver/MotoPlus/CtrlGroup.c#L349) for the section that uses mpSvsGetVelTrqFb. However, the data obtained by that API updates significantly less frequently than the M Register method. So you'll have to choose which method is best for your application.

gavanderhoorn, 2019-03-30:
> I think (I'm not certain) that the method used by DX100 and FS100 does actually indicate direction.

I can confirm it does and I must say I'd assumed there would be feature parity with the M-registers based approach.

Are the M-registers always unsigned?

zhubeilife, 2019-03-30:
Or can we get the speed direction by difference of old position and new one. It seems a dirty way but works.

ted-miller, 2019-04-01:
The value that the system software puts into the M register is unsigned. The best way to determine direction would be to monitor the change in position.

gavanderhoorn, 2019-09-19:
I'm going to close this, as it's not something we can change here.

@ted-miller: would it perhaps be an idea to make MotoROS report velocities without direction sign across all controllers? To improve consistency?

closed 2019-09-19 by gavanderhoorn