ros-industrial/motoman #273: MotoPlus application load error on YRC1000micro controller

xiaodaxia-2008, 2019-03-26:
I compiled the motomandriver code with MotoPlusIDE for YRC1000micro by using the same compile parameters, then I load the *.out application to the controller, but when it starts up, there are 3 alarms: 
ALARM 4207 SYSTEM ERROR(MOTION) [1101], 
ALARM 1020 MOTOPLUS APPLICATION LOAD ERROR [5], 
ALARM 1050 SET-UP PROCESS ERROR(SYSCON) [1].  
However, if i load the MotoRosYRC1u_181.out, there is no alarm at all.  
I searched the ALARM 1020 - 5, it says "Undefined symbols are included in the application", but I didn't revise the source file.
Could you please help about how to compile and build the successful *.out application with these code for YRC1000micro ?

gavanderhoorn, 2019-03-26:
> I compiled the motomandriver code with MotoPlusIDE for YRC1000micro by using the same compile parameters [..]

Could you clarify what you mean by this?

Can you describe your workflow in some more detail? Which project do you load? Which version of MotoPlus do you have? Which version of Windows? Which version of Visual Studio? What is the size of the binary (ie: the `.out` file)? Could you attach the file to this ticket?

xiaodaxia-2008, 2019-03-26:
Thanks for your reply, below is some detailed information.

I referred this http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer tutorial and install the MotoRosYRC1000u.out application in the YRC1000micro, then it is ok with ROS.
Then I get the source files from https://github.com/ros-industrial/motoman/tree/kinetic-devel/motoman_driver/MotoPlus , compiled with MotoPlusIDE for YRC1000micro(Version 1.00), using compiling parameters same as YRC1000uCompilerArguments.mps, then there is .out application. I load this one into YRC1000micro and delete the previous one, after restart the controller, it shows the alarms mentioned before. 
The size of .out file is about 151KB, please to this link https://github.com/xiaodaxia-2008/MotominiDriver . The windows is win10 (64bit) installed on  Oracle VM VirtualBox (the base system is Debian 9). For Visual Studio, I only review code with it ( because it cannot build the MotoPlus application),  the Visual Studio is 2017 community version.

gavanderhoorn, 2019-03-27:
@Tom-Pfahler, @ted-miller: any ideas?

gavanderhoorn, 2019-03-27:
Link to binary posted by @xiaodaxia-2008: [MotominiDriver.out.zip](https://github.com/ros-industrial/motoman/files/3013259/MotominiDriver.out.zip).

ted-miller, 2019-03-27:
I don't see anything in the code which would cause Alarm 1020[5]. Most likely, it's because your ParameterExtraction.yrcmlib is old. Please update that library to the [latest version on github](https://github.com/ros-industrial/motoman/blob/kinetic-devel/motoman_driver/MotoPlus/ParameterExtraction.yrcmlib). Then, recompile the code and try again.

If that doesn't resolve the issue, then the only way to discover the missing symbol is to connect a computer monitor to the robot controller. There is a DisplayPort connector inside the robot controller under the top panel. When you boot up the robot controller, it will show the error message and missing symbol on the display. But, it will scroll quickly off the screen, so you will probably need to record a video of the monitor using your phone.
[image]

xiaodaxia-2008, 2019-03-28:
Thanks for your insturctions, I have solved the problems.
I connected a monitor and found that the GP_xxxxx functions provided by .yrcmlib are unknown symbols.
It seems that the MotoPlus IDE for YRCmicro didn't link the .yrcmlib files to the executables automatically though i have add the libraries to the project. I revised the link parameters manully and got the correct result. 
By the way, could i know some details about the .yrcmlib ( I'll apprecite it if I could have a copy of the source code ) ? Thanks.

closed 2019-03-28 by xiaodaxia-2008

xiaodaxia-2008, 2019-03-28:
The link parameters need to be modified in the MotoPlusIDE or edit  the .yrcmProj file, there is no source code revised. And the .yrcmlib files need to be added to the link parameter **with its absolute paths**, so maybe it's not  convenient to edit here...

gavanderhoorn, 2019-03-28:
> The link parameters need to be modified in the MotoPlusIDE or edit the .yrcmProj file, there is no source code revised. And the .yrcmlib files need to be added to the link parameter **with its absolute paths**, so maybe it's not convenient to edit here...

Do you mean [MpRosAllControllers.vcxproj](https://github.com/ros-industrial/motoman/blob/80c529979de52bb6eed5988d1e3076ae9abc131e/motoman_driver/MotoPlus/MpRosAllControllers.vcxproj)? There is no `.yrcmProj` in the repository (but it could be auto-generated locally).

> there is no source code revised. 

It's still a text file, so it would still be good to keep track of changes you needed to get things to work.

ted-miller, 2019-03-28:
@xiaodaxia-2008 is using MotoPlusIDE, which is a stand-alone development environment (does not use Visual Studio). MotoPlusIDE is currently only available in the Asia region. Although the MotoROS source is 100% compatible with MotoPlusIDE, I don't currently maintain project files in the Github repo.

I am surprised to hear that the .yrcmlib wasn't automatically linked after adding it to the project. I will mention it to my colleagues who maintain MPIDE.

gavanderhoorn, 2019-03-28:
Yes, I was wondering whether that was an important detail in his comments.

> I am surprised to hear that the .yrcmlib wasn't automatically linked after adding it to the project. I will mention it to my colleagues who maintain MPIDE.

would it be an idea to -- in the meantime -- to store `.*Proj` files for MotoPlusIDE here as well?

ted-miller, 2019-03-28:
> By the way, could i know some details about the .yrcmlib ( I'll apprecite it if I could have a copy of the source code ) ? 

Regarding the yrcmlib, this is a closed-source library. It allows the MotoROS application to run on any robot model by reading the internal configuration parameters at runtime. The details of those parameters are confidential. So, I can only distribute the library as a compiled binary.

ted-miller, 2019-03-28:
> would it be an idea to -- in the meantime -- to store .*Proj files for MotoPlusIDE here as well?

I'll add that to my todo list.

xiaodaxia-2008, 2019-03-29:
> Regarding the yrcmlib, this is a closed-source library. It allows the MotoROS application to run on any robot model by reading the internal configuration parameters at runtime. The details of those parameters are confidential. So, I can only distribute the library as a compiled binary.

Ok, thanks any way.

gavanderhoorn, 2019-03-29:
Tracking adding those project files to MotoROS in #275.