ros-industrial/motoman #274: Can Joint_feedback timestamp using Motoman Controller time?

zhubeilife, 2019-03-29:
Hi,

I am wondering can we change the Joint_feedback message's time/timestamp using the controller time? Because it may be more accurate and useful when syncing the robot motion.

gavanderhoorn, 2019-03-29:
> can we change the Joint_feedback message's time/timestamp using the controller time? 

In principle this would be possible, but there is a complicating factor: synchronisation with other ROS message streams that do not use the controller's local clock in that case (ie: everything not published by `motoman_driver`).

The issue is twofold:

 1. clock synchronisation
 1. clock drift

The first can be addressed by either having all ROS node graph participants use a common clock (such as an NTP source), or by manually updating time-stamps in `motoman_driver` messages after an initial calculation of the delta-t between the external PC and the controller.

The second is harder: the clocks of the controller and any external devices may not actually tick at the same frequency, slowly introducing a delta-t between them. This could be addressed by again using a common clock and syncing with it often.

The result of both 1 & 2 is that messages with timestamps coming from the controller would either be "from the past" or "from the future". This quickly becomes a problem as for instance TF relies heavily on time being in-sync in all node graph participants (otherwise we get errors like "lookup would require extrapolation into the future/past").

> Because it may be more accurate and useful when syncing the robot motion.

Can you give an example of what you'd like to sync robot motion with?

gavanderhoorn, 2019-03-29:
What we could do is make this optional behaviour: based on a parameter `robot_state_node.cpp` (for instance) would either use the `time` coming from the controller (whenever it is present in the `simple_message` message) *or* use normal ROS time.

If you'd want this, it would be appreciated if you could submit a PR for that, as personally I will not be able to spend time on this feature.

zhubeilife, 2019-03-30:
One reason I want to use the controller time is [Joint velocity feedback value's sign don't change ](https://github.com/ros-industrial/motoman/issues/272). We are thinking to use the controller time to do the difference to get the speed. If using PC time, we don't know the real duration between the two points.

gavanderhoorn, 2019-03-31:
Wouldn't checking whether current joint angles are `>` or `<` than the ones from the previous cycle in MotoROS be easier and more reliable? That wouldn't need any changes to the published messages.

gavanderhoorn, 2019-09-04:
I'm going to close this due to inactivity.

If you're still interested in this @zhubeilife, it might be most expedient to submit a PR to get the desired functionality merged into MotoROS.

closed 2019-09-04 by gavanderhoorn