ros-industrial/motoman #276: tool0 frame not at end of link_6_t in motoman_mh50_support

npd26, 2019-05-27:
Hello,

this is issue regarding my question on ROS answers regarding motoman_mh50_support package.

> https://answers.ros.org/question/324031/motoman_mh50_support-urdf-frame-location/?comment=324060#post-id-324060

why is there some gap? it should be exactly at the end, or?

Thanks.

gavanderhoorn, 2019-05-27:
@EricMarcil: would you have any idea? Is this a (small) issue with the origin of the mesh for the T link?

EricMarcil, 2019-06-04:
It is an issue with the origin of the mesh but not just the T-axis.  It's the mesh of the whole upper arm that seem to be shifted backward by about 13 mm.  It's a bit hard to eyeball the proper correction.  I've request new model files from my design department and will try to generate proper mesh from them.

npd26, 2019-06-05:
can this cause problem to vision based pick of object?

gavanderhoorn, 2019-06-05:
I believe so: if you've attached your EEF to the urdf at that link then FK and IK would probably not work correctly.

npd26, 2019-06-05:
I have attached my realsense camera URDF directly to tool0. I detect object center (x,y) in camera_frame and then directly transforming it to base_link. since 2 week I am trying and tool is not reaching to correct position. can I consider this as one reason for that?

gavanderhoorn, 2019-06-05:
Any vision process is going to have (quite) a bit of error estimating poses, but if your kinematic model is incorrect then anything using that model will also include that error.

npd26, 2019-06-05:
ok I will check it. I hope we will get new support package quickly.

EricMarcil, 2019-06-05:
I don't agree.  The problem should be strictly visual.  If you are attaching your tool to the Tool0, then it should resolve properly.  As a reference, when the robot is at Home (all angles at 0) the Tool0 position should be: 
X: 1345  Y: 0  Z:1080 (or Z: 1620 relative to the ground level).

gavanderhoorn, 2019-06-05:
@EricMarcil wrote:
> I don't agree.

and I agree with your disagreement.

@npd26: I'd misunderstood the issue.

if the kinematic structure is correctly encoded in the xacro macro, and it would appear so, then the fact that the mesh itself is not properly aligned with that should not affect motion planning or IK/FK.

gavanderhoorn, 2019-06-05:
> As a reference, when the robot is at Home (all angles at 0) the Tool0 position should be:
> X: 1345 Y: 0 Z:1080 (or Z: 1620 relative to the ground level).

@npd26: you should be able to verify this by running:

```
rosrun tf tf_echo base tool0
```

the translation reported should correspond to what @EricMarcil specified.

npd26, 2019-06-05:
ok now understand. so error may be because I have manually given camera joint position w.r.t tool0. I will check it.

npd26, 2019-06-06:
I have verified it. it is same on tool0 position is same on teach pendant and in TF.

gavanderhoorn, 2019-06-06:
Then the problem would seem to be merely cosmetic and your problems picking things up would seem to come from somewhere else.

npd26, 2019-06-06:
yes, it seems only visualization problem.

gavanderhoorn, 2019-06-06:
Let's keep this open until @EricMarcil has had a chance to fix it.

EricMarcil, 2019-06-12:
I've got a new model but it has the same problem.  The L-axis has some rotation in the model and that is moving the upper arm backward.  It's have to correct without knowing the exact angle.  I've requesting that the original model be moved to the proper home position and regenerated.  Once, I have the corrected model, I will update the mesh models.

EricMarcil, 2019-06-14:
@gavanderhoorn : I've made a pull request #279 but I don't think that I've done it correctly.  There seems to be a bunch of commit related to the Mini that when along with it.  It should just be the last commit MH50 meshes correction that should be included.  Can you check on it.  What is the best way to fix it?

EricMarcil, 2019-06-14:
New meshes were pushed to: ros-industrial/motoman

gavanderhoorn, 2019-06-14:
@EricMarcil: you still need to submit a new PR :)

npd26, 2019-06-24:
will this merging performed automatically?

gavanderhoorn, 2019-06-24:
No, the PR will have to be review first.

You don't need to wait on that though, it's perfectly possible to use a PR locally, see [Checking out pull requests locally](https://help.github.com/en/articles/checking-out-pull-requests-locally) for instance.

EricMarcil, 2020-02-27:
Sorry, it took so long.  I finally got a model that lined up correctly and was able to generate new meshes from it.  We should be able to close this issue once the review of PR #316 is done.

gavanderhoorn, 2020-10-05:
#316 was supposed to fix this, but it got closed -- I believe by accident.

If we can get #316 re-opened and merged, that would fix this issue.

gavanderhoorn, 2020-10-06:
#357 will now fix this.

PR #357 (2020-10-06): Resubmit of 316: MH50 meshes update
This is #316, as that never got merged.

It's the exact same commit by @EricMarcil, but now in a new branch and committed today (but authored on the original date).

This should fix #276.
closed 2020-10-06 by gavanderhoorn