ros-industrial/motoman #278: Update wiki to list YRC micro support

gavanderhoorn, 2019-06-06:
The current version of [wiki/motoman_driver](http://wiki.ros.org/action/recall/motoman_driver?action=recall&rev=31) doesn't *explicitly* list the YRC 1000 micro controller as supported.

It does say:

> The software will work on all FS100, DX100, DX200, and YRC1000 robot controllers.

but that is causing confusion (https://github.com/ros-industrial/motoman/issues/249#issuecomment-499293841).

The YRC 1000 micro should be listed as fully supported, including which system software version it should have minimally (under the *Controller system software* section).

gavanderhoorn, 2019-06-06:
I've assigned this to you @ted-miller as I don't know which system version YRC micros would need for MotoROS.

gavanderhoorn, 2019-06-06:
Note that the [wiki/motoman_driver/Tutorials/indigo/InstallServer](http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer) page does list the YRC micro *and* has a link to the specific MotoROS version to be used [here](http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer#Installation_from_Binary_.28MotoROS_Driver_v1.5.0_and_higher.29).

gavanderhoorn, 2019-06-06:
@ted-miller: I've added what I know to the page ([diff](http://wiki.ros.org/action/info/motoman_driver?action=diff&rev2=32&rev1=31)).

I couldn't find the order nr for MotoPlus for the micro (don't even know whether it's different from the regular micro).

If you could check the page and add/update where necessary?

ted-miller, 2019-06-07:
I have updated everything except for the EU part number. I've reached out to my colleagues and update the number when they respond.

ted-miller, 2019-06-11:
The EU part number (206078) for the MotoPlus SDK has been updated on the wiki.

closed 2019-06-11 by ted-miller

gavanderhoorn, 2019-06-11:
Thanks @ted-miller for all the corrections. Much appreciated :+1: