ros-industrial/motoman #282: mh50_support urdf:  incorrect link_3_u joint limits?

cjue, 2019-07-05:
Hi,

it looks to me like there is an error for one of the joint limits in the MH50 URDF: https://github.com/ros-industrial/motoman/blob/kinetic-devel/motoman_mh50_support/urdf/mh50_macro.xacro#L125
```
<limit lower="-1.39626" upper="3.59538" effort="2745.86" velocity="3.106"/>
```

The specification of the MH-50 II here says the limits are +251/-170°, not +205/-80°:

https://www.motoman.com/en-us/products/robots/industrial/assembly-handling/mh-series/mh50-ii

Is this an error, or was this correct for an earlier MH50 variant?

gavanderhoorn, 2019-07-05:
Afaik, the MH-50 II and the model supported by `motoman_mh50_support` are not necessarily the same robot.

I'm not entirely sure though, so I'll wait for @EricMarcil to confirm this.

EricMarcil, 2019-07-11:
The values in the model are correct.  They are defined relative to the previous joint (L-axis).
The specification you are making reference to is the possible orientation of the U-axis relative to the ground which also depends on the L-axis possible position, therefore yielding to a larger range.

closed 2019-07-11 by EricMarcil

MuhongGuo, 2019-11-12:
@EricMarcil @cjue I think the reason is that it is impossible for the U-axis to reach lower degree than -80 because of physical interference from the motor on L-axis. Please reference to the image below, where the U-axis is at the position of -90 degree and it is already in collision. I've also tried to change the position of L-axis but couldn't find any condition to let U-axis reach the range as per the specs.

[image]

EricMarcil, 2019-11-12:
@MuhongGuo : I wouldn't call it unrealistic, it's just a different reference than what you were expecting.  Like I said, that reference is relative to the ground or the horizontal line.  If the L-axis is at -90 deg and the U-axis -80 deg relative to the L-axis, that puts the U-axis at -170 deg relative to the horizontal.  Marketing people don't always think like engineers...  That being said, the specifications for the new series like the GP series, are represented as you would expect it in reference to the previous link.

MuhongGuo, 2019-11-12:
> If the L-axis is at -90 deg and the U-axis -80 deg relative to the L-axis, that puts the U-axis at -170 deg relative to the horizontal. 
@EricMarcil By the -90 deg, are you referring to the lower limit of L-axis? But the rotation axis of L-axis is opposite to U-axis's. If the lower limit of U-axis in the spec is w.r.t the ground, it should be -(130 + 80) = -210 (deg).

EricMarcil, 2019-11-12:
OK, you're right I might have did that addition too quickly.  Here are where those angles are coming from:

[image]

[image]

EricMarcil, 2019-11-12:
@MuhongGuo : See above

MuhongGuo, 2019-11-13:
@EricMarcil Could you please tell me the positions of u-axis in these two configurations of the arm?

EricMarcil, 2019-11-13:
@MuhongGuo: I'm not the one that does those calculations for the bulletin, but I think the reference might be the vertical, when considering the L-axis values.  So the top one I think is 251 deg and the bottom one -170 from the vertical.  In any event, the values in the URDF are matching the ones from the parameter file from the controller for that robot.  
Note that there are also some mechanical restrictions between L-U axes that can occur depending where you are in space to prevent colliding with other part of the robot like the base.  So the position  you are showing above in yellow can't be reached on the real robot.  You'll get a "Special Machanical Interference L-U" message on the pendant.  I don't know of a good way of capturing that within the URDF definition.