ros-industrial/motoman #287: motion control Start Traj Mode can return 'success' when e-stop is pressed

gavanderhoorn, 2019-09-04:
Something I've observed today: the `Motion Motion Control`service request (at the simple message level, packet id: `0x7d1`) sometimes appears to return a `Reply Code` with value `1` (which would denote `Success`) even if/when the teach pendant e-stop is pressed.

Wireshark dissection of the request:

```
ROS-Industrial SimpleMessage, Motoman Motion Ctrl (0x7d1), 68 bytes, little-endian
    Prefix
        Packet Length: 64
    Header
        Message Type: Motoman Motion Ctrl (0x000007d1)
        Communications Type: Service Request (2)
        Reply Code: Unused / Invalid (0)
    Body
        Robot ID: 0
        Sequence Number: 0
        Command: Start Traj Mode (200121)
        Data (reserved)
            J0:    0,000000000
            J1:    0,000000000
            J2:    0,000000000
            J3:    0,000000000
            J4:    0,000000000
            J5:    0,000000000
            J6:    0,000000000
            J7:    0,000000000
            J8:    0,000000000
            J9:    0,000000000
```

and the reply:

```
ROS-Industrial SimpleMessage, Motoman Motion Reply (0x7d2), 76 bytes, little-endian
    Prefix
        Packet Length: 72
    Header
        Message Type: Motoman Motion Reply (0x000007d2)
        Communications Type: Service Reply (3)
        Reply Code: Success (1)
    Body
        Robot ID: -1
        Sequence Number: -1
        Command: Unknown (2001)
        Result: Success/True (0)
        Subcode: Unknown (0)
        Data (reserved)
            J0:    0,000000000
            J1:    0,000000000
            J2:    0,000000000
            J3:    0,000000000
            J4:    0,000000000
            J5:    0,000000000
            J6:    0,000000000
            J7:    0,000000000
            J8:    0,000000000
            J9:    0,000000000
```

Note also how the `robot_status` topic claims that "motion is possible", even though the e-stop is pressed and the `e_stopped` field is also high:

```
ROS-Industrial SimpleMessage, Status (0x0d), 44 bytes, little-endian
    Prefix
        Packet Length: 40
    Header
        Message Type: Status (0x0000000d)
        Communications Type: Topic (1)
        Reply Code: Unused / Invalid (0)
    Body
        Drives Powered : False (0)
        E-Stopped      : True (1)
        Error Code     : 0
        In Error       : False (0)
        In Motion      : False (0)
        Mode           : Auto (2)
        Motion Possible: True (1)
```

gavanderhoorn, 2019-09-04:
Behaviour described in the OP was observed on a YRC1000 with MotoROS `v1.9.0` (testing #284).

I'll downgrade to `v1.8.2` to see if this behaviour is also present in that version.

gavanderhoorn, 2019-09-04:
@ted-miller: any idea?

gavanderhoorn, 2019-09-04:
On this particular system there is no external e-stop, so I cannot test whether that changes anything, but I don't expect it to.

ted-miller, 2019-09-04:
I see the problem. Ros_Controller_StatusUpdate (Controller.c) doesn't include estop in the decision logic.

I'll work on it this afternoon and push to #284 .

ted-miller, 2019-09-05:
This should be fixed in af3a9c95da5fe7001d7c74562d81b7ee7e56e6b5
Please verify.

gavanderhoorn, 2019-09-05:
Seems to have been fixed. Thanks :+1: .

I'll leave this open until #284 has been merged.

gavanderhoorn, 2019-11-14:
#284 was merged, so closing.

closed 2019-11-14 by gavanderhoorn