ros-industrial/motoman #288: Support Request for YRC1000 GP88

EricMarcil, 2019-09-04:
From #277  
I am currently working on basic GP88 support and would like to add the correct efforts there as well. Are these YRC1000 parameter charts available for download anywhere, or could you post the values for the GP88?

EricMarcil, 2019-09-04:
I didn't want to post the GP88 information on #277 which is related to the GP180 to avoid any confusion.  The requested information is below:
[image]

cjue, 2019-09-05:
Thank you, now I only need to fix my collision meshes for my GP88 PR.

gavanderhoorn, 2019-11-05:
I'll close this as we have an open PR that adds support and the (initial) question by @cjue has been answered.

closed 2019-11-05 by gavanderhoorn