ros-industrial/motoman #291: Note that the [wiki/motoman_driver/Tutorials/indigo/InstallServer](http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer) page does list the YRC micro *and* has a link to the specific MotoROS version to be used [here](http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer#Installation_from_Binary_.28MotoROS_Driver_v1.5.0_and_higher.29).

Gloriabhsfer, 2019-09-11:
Note that the [wiki/motoman_driver/Tutorials/indigo/InstallServer](http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer) page does list the YRC micro *and* has a link to the specific MotoROS version to be used [here](http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer#Installation_from_Binary_.28MotoROS_Driver_v1.5.0_and_higher.29).

_Originally posted by @gavanderhoorn in https://github.com/ros-industrial/motoman/issues/278#issuecomment-499375268_

gavanderhoorn, 2019-09-11:
I would suggest to follow the [installation tutorial](http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer), after having checked [motoman_driver](http://wiki.ros.org/motoman_driver) for the prerequisites, such as required versions of the controller's system software.

The YRC1000 micro is fully supported and also listed and mentioned on the pages I linked to.

If anything is unclear about that, you're welcome to post a comment on this issue again.

gavanderhoorn, 2019-09-11:
As this is not an issue with the packages in this repository, I'll close the issue.

If it turns out it actually is, we can re-open.

In the meantime, feel free to keep commenting of course.

closed 2019-09-11 by gavanderhoorn

Gloriabhsfer, 2019-09-11:
> I would suggest to follow the [installation tutorial](http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer), after having checked [motoman_driver](http://wiki.ros.org/motoman_driver) [...]

Thanks for your reply.
We follow the installation steps on the website but our controller cannot recognize the .out file. Do you have any ideas about that?
Thank you
Gloria

EricMarcil, 2019-09-11:
There are two pendants available for the YRC1000Micro.
Do you have the standard pendant with a small screen and a keypad or do you have the Smart Pendant with a large screen (like a tablet)?

Gloriabhsfer, 2019-09-11:
> There are two pendants available for the YRC1000Micro.
> Do you have the standard pendant with a small screen and a keypad or do you have the Smart Pendant with a large screen (like a tablet)?

Yes, we use the smart pendant now. I think that's the problem that smart pendant cannot find the .out format. But I have no idea about how to deal with this issue.

EricMarcil, 2019-09-11:
The Smart Pendant doesn't support the management of MotoPlus application (it's currently in development).
You will need to use the "Software Pendant" which is a version of the standard pendant that you can run on a Windows PC.  You can get a copy from the Smart Pendant.  Connect a USB drive to the Smart Pendant and then select "System Settings" --> "General".
Down at the bottom of the screen in the "Bundled Resources" select the "Software Pendant" Application and then press the "Export..." button.
I believe you'll get the Manual and Installation files.  
Once that is installed, you can connect to the PC to the Controller and boot-up in Maintenance mode and follow the instructions from the Wiki.

Gloriabhsfer, 2019-09-11:
> The Smart Pendant doesn't support the management of MotoPlus application (it's currently in development).
> You will need to use the "Software Pendant" which is a version of the standard pendant that [...]

Thank you. Not we are in maintenance mode and we can install the file with ROS Wiki. I have really appreciated your works on Smart Pendant. Please let me know when it will finish.
Thanks again.
Gloria

Gloriabhsfer, 2019-09-11:
> The Smart Pendant doesn't support the management of MotoPlus application (it's currently in development).
> You will need to use the "Software Pendant" which is a version of the standard pendant that [...]

Thank you again for all your help! Referencing above we have successfully booted into maintenance mode using the Yaskawa software Pendant  for YRC on our PC. To do this we have followed the directions set forth in Yaskawa's YRC1000micro OPTIONS INSTRUCTIONS: for soft pendant (US part #181303-1cd, Japanese Part #HW1484787) pgs 38-39. Where we run into trouble is loading the MotoRosYRC1000u.out file into the SoftwarePendant for YRC program on our PC. Following the Motoman tutorial (Installing the Motoman ROS Server) we are unable to see the file in the SoftwarePendant for YRC program. Specifically, following the steps outline under "FS100,DX200, and YRC", we are stopped at step 6 where we are unable to load MotoRosYRC1000u.out. We instead receive "Error 3090 Transmission error with the media [0]" message.

Page 35 if the Yaskawa's YRC1000micro OPTIONS INSTRUCTIONS: for soft pendant manual explains where the SoftwarePendant for YRC looks to for External Memory. Stating they are saved in the PC-HDD folder under the SWP folder in Documents. In an attempt to load the binary file, we copied and pasted the MotoRosYRC1000u.out folder to the PC-HDD folder. However, still can not see it in our SoftwarePendant for YRC program...

Any suggestions?

Thank you again for all your help with this!

benoit0192, 2022-04-27:
Hi all and thank you for your good work.
I am not sure if this is the good place or if this issue already came up before so please let me know.
I am encountering an issue regarding the Software pendant interface with respect to the screenshots from the documentation.

Specifically, I would like to install MotoROS using the binary files but as you can see with the attached file below, when it comes to using the software pendant in "Maintenance Mode", the "MotoPlus APL" tab on the left side does not show up (sorry, the interface is in Japanese). Because of that, I cannot complete the instructions from http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer and load the ".out" binary file.

FYI, here is my setup:
- Motoman HC10DT
- YCR1000micro controller (YBS2.34.00A(JP/EN)-00)
- Tablet-like teaching pendant

I got the software pendant by exporting it from physical teaching pendant as instructed. Based on the version I am using and the minimum version mentions in the instructions (YBS2.30.00-00), it looks like there should be no problem.

Regardless, would there be an issue with the version I am using or did the installation procedure changed since then? Or do I need to purchase the MotoPlus SDK? Or else..

Thank you in advance for your feedback.

[image]

gavanderhoorn, 2022-04-27:
> FiXED: I figured I add to check the "MotoPlus" option in the "System" menu. I did so and now the "MotoPlus" does show up as expected so there is no more issue.

Good to hear you figured it out.

@EricMarcil / @ted-miller: would this be something to add to the tutorial?

ted-miller, 2022-04-27:
@benoit0192 :
> sorry, the interface is in Japanese

You can toggle the display language by clicking [SHIFT] then [AREA] on the virtual keypad.
[image]

@gavanderhoorn:
> would this be something to add to the tutorial?

This is covered in [section 2.4.2](http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer#FS100.2C_DX200.2C_and_YRC). I suspect it was simply overlooked.

benoit0192, 2022-04-28:
@ted-miller 

> > sorry, the interface is in Japanese
> 
> You can toggle the display language by clicking [SHIFT] then [AREA] on the virtual keypad.

Yes I read this on other issues and tried it out but nothing changes so I thought there was no english support on the version I am using.

> > would this be something to add to the tutorial?
>
> This is covered in [section 2.4.2](http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer#FS100.2C_DX200.2C_and_YRC). I suspect it was simply overlooked.

Yes indeed it was clearly mentioned. I simply overlooked.