ros-industrial/motoman #292: Question about motion_possible in YRC1000 micro controller

Gloriabhsfer, 2019-09-13:
Hi all:
Thanks for your help about #291 and now we established the connection through our PC and GP8 robot. 
Now we want to control the joint through ROS. We run the move_to_joint.py and the robot won't move. So we checked through the internet and call rosservice and get feedback error like this:
[image]
(Sorry this is my labmate's computer)
It seems like we enable the robot but the motion_possible is not on, so when we set joint information to robot, it won't work.

Any ideas?
Thanks again for your help
Gloria

EricMarcil, 2019-09-13:
You can refer to the document at: [http://wiki.ros.org/motoman_driver?action=AttachFile&do=view&target=MotoRos_EDS.pdf](url)
Chapter 4: Appendix gives the various return value meaning.
5008 Corresponds to ROS_RESULT_NOT_READY_NOT_STARTED

Basically, the robot is not in playing back the ROS_INIT job.   In normal operation the green light on the teach pendant should be on if the job is playback.

To troubleshoot that the best way is to manual try to start the INIT_ROS job on the robot.  On the Pendant switch to Play mode, select the ROS_INIT job, turn on the servos and start the job.  You'll probably get an message during that process that will indicate what's wrong.

Gloriabhsfer, 2019-09-14:
> You can refer to the document at: [http://wiki.ros.org/motoman_driver?action=AttachFile&do=view&target=MotoRos_EDS.pdf](url)
> Chapter 4: Appendix gives the various return value meaning.
> 5008 Corresponds to ROS_RESULT_NOT_READY_NOT_STARTED
> 
Sorry I cannot open this link. Is it in ROS wiki for Motoman? I think that's a good reference I can read but I cannot find it online.
> Basically, the robot is not in playing back the ROS_INIT job. In normal operation the green light on the teach pendant should be on if the job is playback.
>
> To troubleshoot that the best way is to [...]
I think want you mentioned make sense, I need to start the ROS_INIT job. But as I mentioned in another issue, I only have the smart pendant now and I am now familer with it. Can I start the INIT_ROS job on smart pendant? Or I need the standard pendant on PC and run that job.

And If I use the ROS_INIT job ( with MoveIt or just give it some simple ros_control command) it will gave me error all the time but I can ignore it?

Thanks for your quick reply again,
Best
Gloria

gavanderhoorn, 2019-09-15:
@Gloriabhsfer: I've fixed the link.

Bdelspi, 2019-09-15:
@gavanderhoorn Thank you for all your help with this! @Gloriabhsfer and I are working on this together and appreciate all your help.

Background info: We do not have the standard programming pendant, but rather have the smart pendant, and are thus using the smart pendant and software pendant on windows PC.

Implementing 1 of 2 setups:
Setup1) Smart pendant in remote mode, running the software pendant on our windows PC in remote mode
Setup2) Smart pendant in play mode, running software pendant on our PC in play mode

Referring to the comment _"Basically, the robot is not in playing back the ROS_INIT job. In normal operation the green light on the teach pendant should be on if the job is playback."_

The smart pendant reads in the lower right hand corner ROS_INIT, suggesting we are playing back the ROS_INIT job. Watching the software pendant on windows PC, but sending commands to the controller via ROS, we are able to confirm servo power on via sound (click upon turning servo power on) and green light lit on the software pendant.

We get no error messages however, when we run the move_to_joint.py command, no link motion results...

Suggestions?

> @Gloriabhsfer: I've fixed the link.

gavanderhoorn, 2019-09-16:
Can you please describe the *exact* sequence of commands (including opening windows, clicking on buttons/things, etc), starting programs, ROS nodes, launch files, etc that lead to the error you report.

If possible, please also capture a trace of the network traffic using [wireshark](https://www.wireshark.org/). That would let us verify whether the problem is on the controller side or on the ROS side.

EricMarcil, 2019-09-16:
On the pendant, not using ROS, are you able to select the INIT_ROS job and play it back to where it should get to the WAIT line in the job?

Bdelspi, 2019-09-16:
@gavanderhoorn & @EricMarcil Thank you so much for all your help with this! Our home position was misaligned, after re-calibrating our home position we were able to run the move_to_joint.py command flawlessly!

Thank you both again!

gavanderhoorn, 2019-09-19:
It seems the issues reported here have been fixed, or at least it would appear that you can now control the robot from ROS.

I'm not entirely sure how the home position plays into this, but that's not too important.

Closing.

gavanderhoorn, 2019-09-19:
@EricMarcil / @ted-miller: should we log an issue to make sure we add some more clarification for the Smart Pendant (what's in a name) to the installation tutorial?

closed 2019-09-19 by gavanderhoorn

Gloriabhsfer, 2019-09-19:
Sorry for my late reply. Yes, now we can send the command through motoman_driver to control the robot. But we still deal with the MoveIt communication to YRC1000 controller. So I just keep the issue on and didn't close it. But thanks for your help again.

Bdelspi, 2019-09-23:
> @EricMarcil / @ted-miller: should we log an issue to make sure we add some more clarification for the Smart Pendant (what's in a name) to the installation tutorial?

Yes please, talking with Yaskawa their exact words when I asked the difference between the SmartPendant and Standard Pendant were "night and day". The software Pendant helps, but also does not have the full functionality of the standard pendant...

gavanderhoorn, 2019-09-23:
This is already being tracked in #293.

ted-miller, 2019-10-24:
The installation tutorial has been updated to include Smart Pendant.