ros-industrial/motoman #293: Update installation instructions for regular/Smart pendant

ted-miller, 2019-09-20:
- Add clarification that current instructions are for regular pendant
- Add additional instructions for Smart Pendant.

gavanderhoorn, 2019-09-20:
Context: #292 and #291.

ted-miller, 2019-10-24:
The installation tutorial has been updated to include Smart Pendant.

closed 2019-10-24 by ted-miller

gavanderhoorn, 2019-10-24:
Diff: [here](http://wiki.ros.org/action/diff/motoman_driver/Tutorials/indigo/InstallServer?action=diff&rev1=44&rev2=46).

gavanderhoorn, 2019-10-24:
@ted-miller: I noticed that the text you added says this at some point (step 6 of section 2.1):

> Refer to chapter 12 of this manual for the installation and usage of the PC Software Pendant. 

There is no "chapter 12". Was this copied from a Yaskawa Motoman manual?

gavanderhoorn, 2019-10-24:
Ah, I believe I understand now: "the manual" refers to:

> open the “documentation-index.html” file and then select the “Instructions for Smart Pendant (YRC1000/YRC1000micro)”

ted-miller, 2019-10-24:
Updated for clarity:

> In the Pendant/Documentation folder, open the “documentation-index.html” file and then select the “Instructions for Smart Pendant (YRC1000/YRC1000micro)”
Refer to chapter 12 of “Instructions for Smart Pendant (YRC1000/YRC1000micro)” for the installation and usage of the PC Software Pendant

gavanderhoorn, 2019-10-24:
I've made some minor formatting fixes as well: [diff](http://wiki.ros.org/action/info/motoman_driver/Tutorials/indigo/InstallServer?action=diff&rev2=48&rev1=47).