ros-industrial/motoman #294: Question of move_to_joint.py in Motoman_driver

Gloriabhsfer, 2019-09-25:
Hi all,

I am planning to set groups of waypoints for my Motoman GP8 robot (YRC1000 Mirco with smart pendant). And I just start look at the python file.
In that python code, what you get is form a rosbag
https://github.com/ros-industrial/motoman/blob/197d2f8da805c9849a821e8b627748811b3242a1/motoman_driver/src/move_to_joint.py#L80-L89
And I have read the rosbag tutorial about how to record a rosbag.[http://wiki.ros.org/rosbag/Tutorials/Recording%20and%20playing%20back%20data](url)
But in this code, you set waypoint when you run the python script, and you didn't generate or save the joint in a rosbag file. Cause what I want to do is set a group of waypoint and let the robot following these points. In my understanding, I need to set the directory with where you set the rosbag and run the waypoint in rosbag step by step. But I am not sure it's right or not.
Please correct me if my understanding is wrong. 

Thanks,
Gloria

gavanderhoorn, 2019-09-25:
I would refer you to #229. That links to my work where I've a skeleton of an action server client that is probably what you are looking for.

gavanderhoorn, 2019-09-25:
As I've noted in https://github.com/ros-industrial/motoman/issues/229#issuecomment-402759642, that script was not tested too much, so you may run into some minor issues.

I see now that I've not actually (deep) cloned the trajectory pts, so no motion may result (or only a single motion, instead of three).

The approach shown would still be valid though.

Gloriabhsfer, 2019-09-25:
Thanks for your reply, the code in #229 seems works. Yes, I agree we can use some motion planning framework and we are now using MoveIt. But the connection has some issue and it seems we didn't set the controller right. So we need to read more about how to set the MoveIt controller.
Now we want first do a demo pick and place task so we want to hard coding it into a python file.
I will close the issue and reply under it if we make any progress.
Best,
Gloria

closed 2019-09-25 by Gloriabhsfer