ros-industrial/motoman #296: How to set the velocity of each joint independently?

kctoayo88, 2019-10-03:
Hello, 
I use motoman driver and move_group to control Motoman MH5 and FS100. 
Recently, I have a project needs to control the velocity of each joint, and I found this API `setMaxVelocityScalingFactor` to change the the max velocity scaling for the all of joints, not independently.
So, I would like to know is there any C++ or Python API can set up the velocity of each joint independently. If not, does it have any possible way to achieve this?

gavanderhoorn, 2019-10-03:
> I have a project needs to control the velocity of each joint, [..]

Can you clarify what you mean by this exactly?

Are you looking to use *velocity control* with a Motoman robot?

Or are you trying to set maximum velocities to be used in trajectory generation?

The former is currently not supported, the latter could be, but would probably be done using joint limits.

> I found this API `setMaxVelocityScalingFactor` to change the the max velocity scaling for the all of joints, not independently.

Please note that this is a function in MoveIt, which is not part of the driver provided here.

We cannot help you with MoveIt.

gavanderhoorn, 2019-10-03:
As this is a question, and not an actionable issue for this repository, I'm going to close it.

Feel free to keep commenting here of course.

If it turns out that this is something that needs changes in this repository, we can re-open.

closed 2019-10-03 by gavanderhoorn