ros-industrial/motoman #297: Did the driver support joint velocity control or joint torque control?

namuhevil, 2019-10-06:
I want to know which interface is the mostly opened gate to access the Yaskawa manipulator, We want to control every joint velocity and torque to do visual servoing.

gavanderhoorn, 2019-10-06:
Only position control is supported.

And to be more specific: the driver (ROS side) expects full trajectories, not single traj pts.

#88 (updated in #215) adds a sort-of position-based streaming interface.

closed 2019-10-06 by gavanderhoorn

harumo11, 2019-10-17:
Hello developers of motoman package and thanks for the great ROS package.

I have a question about point-streaming interface.
I tried to use the point streaming interface, but I couldn't figure out how to use it.

I tried the following steps.

1. `roslaunch motoman_sia20d_moveit_config moveit_planning_execution.launch robot_ip:=10.0.0.2 controller:=fs100 sim:=false`

2. Joint values were sent to `joint_streaming_interface` using [this program](https://github.com/harumo11/sia20/blob/master/sia20_control/src/sequential_ik_solver.cpp)

3. I got the following error when I sent joint values.

I sent  *motion_streaming_interface* joint values
```
[ERROR] [1571296092.465539508]: Trajectory splicing not yet implemented, stopping current motion.
[ERROR] [1571296092.720267573]: Trajectory splicing not yet implemented, stopping current motion.
[ERROR] [1571296092.992569231]: Trajectory splicing not yet implemented, stopping current motion.
[ERROR] [1571296092.998145363]: Aborting Trajectory.  Failed to send point (#0): Invalid message (3) : Trajectory start position doesn't match current robot position (3011)
[ERROR] [1571296093.253868287]: Trajectory splicing not yet implemented, stopping current motion.
[ERROR] [1571296093.509321453]: Trajectory splicing not yet implemented, stopping current motion.
[ERROR] [1571296093.781523688]: Trajectory splicing not yet implemented, stopping current motion.
```

Is there any available implementation example to use point-streaming or documents?

Thanks in advance.

gavanderhoorn, 2019-10-17:
Streaming will only work if you've merged the changes in one of the PRs I linked.

Can you confirm whether you have done that?

harumo11, 2019-10-17:
@gavanderhoorn  Thanks for quick response.

I checked merge log and I can't found merge log of #215 .
I cloned kinetic-devel branch from github and I use it.

Which branch should I use?

gavanderhoorn, 2019-10-17:
You will need to merge / port the changes in either #215 or any of the updated PRs.

Without that this will not work.

harumo11, 2019-10-17:
Sorry but could you please tell me the method for merging?
I thought it is only can be done by someone who is authorized.

gavanderhoorn, 2019-10-17:
You can merge pull requests locally.

[This page](https://help.github.com/en/articles/checking-out-pull-requests-locally#modifying-an-inactive-pull-request-locally) gives some info on how to do these sorts of things.

choonyip, 2019-10-18:
Hello dear @gavanderhoorn, I am a member of the same research group with @harumo11.
I have merged PR #215 based on the page you provided, with the steps below:

```
$ git fetch origin pull/215/head:point_streaming
$ git merge point_streaming
```
```
Merge made by the 'recursive' strategy.
 .../joint_trajectory_interface.h                   |  11 ++
 .../joint_trajectory_streamer.h                    |  10 +-
 .../motoman_driver/joint_trajectory_streamer.h     |   7 +
 .../joint_trajectory_interface.cpp                 |   3 +-
 .../joint_trajectory_streamer.cpp                  | 208 ++++++++++++++++++++-
 motoman_driver/src/joint_trajectory_streamer.cpp   |  83 +++++++-
 6 files changed, 309 insertions(+), 13 deletions(-)
```

But I'm still getting the same error shown above.
Appreciate if you could take a look to our source code, or is there any available implementation example?

gavanderhoorn, 2019-10-18:
Looking at [your program](https://github.com/harumo11/sia20/blob/29a5b807f8e7b04f5f16faf87c59a379ab7c8ee5/sia20_control/src/sequential_ik_solver.cpp), you appear to be sending messages to the `command` topic ([here](https://github.com/harumo11/sia20/blob/29a5b807f8e7b04f5f16faf87c59a379ab7c8ee5/sia20_control/src/sequential_ik_solver.cpp#L38)):

```c++
ros::Publisher joint_trajectory_publisher = node_handle.advertise<trajectory_msgs::JointTrajectory>("/sia20/sia20_joint_controller/command", 1);
```

the correct topic would appear to be `joint_command`. See [here](https://github.com/ros-industrial/motoman/pull/215/files#diff-5e3c36db0a99e03f8d711c4d1632d37bR129).

choonyip, 2019-10-20:
It worked! As you said, we should send messages to /joint _command topic. Thank you so much. Really appreciate your help.

choonyip, 2019-12-14:
@gavanderhoorn Sorry to ask it here, but is there a chance that Universal Robots driver supports position-based streaming interface?

gavanderhoorn, 2019-12-15:
This is the issue tracker for `motoman` and related packages. Please don't ask questions about other packages here.